imyller/meta-nodejs #3: 'virtual/node' breaks for do_package_deb because of '/'

fladi, 2015-03-11:
The char `/` is not allowed in the `Provides:` field in `debian/control`. Perhaps another `PROVIDES` would be useful to make it work with `PACKAGE_CLASSES ?= "package_deb"`.

    | DEBUG: Executing python function do_package_deb
    | dpkg-deb: building package `nodejs-dbg' in `.../tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/nodejs/0.12.0-r1.0/deploy-debs/cortexa9hf-vfp-neon/nodejs-dbg_0.12.0-r1.0_armel.deb'.
    | NOTE: Not creating empty archive for nodejs-staticdev-0.12.0-r1.0
    | dpkg-deb: building package `nodejs-dev' in `.../tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/nodejs/0.12.0-r1.0/deploy-debs/cortexa9hf-vfp-neon/nodejs-dev_0.12.0-r1.0_armel.deb'.
    | dpkg-deb: building package `nodejs-doc' in `.../tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/nodejs/0.12.0-r1.0/deploy-debs/cortexa9hf-vfp-neon/nodejs-doc_0.12.0-r1.0_armel.deb'.
    | NOTE: Not creating empty archive for nodejs-locale-0.12.0-r1.0
    | dpkg-deb: error: parsing file '.../tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/nodejs/0.12.0-r1.0/packages-split/nodejs/DEBIAN/control' near line 14 package 'nodejs':
    |  `Provides' field, invalid package name `virtual/node': character `/' not allowed (only letters, digits and characters `-+._')
    | DEBUG: Python function do_package_deb finished
    | DEBUG: Python function do_package_write_deb finished
    | ERROR: Function failed: dpkg-deb execution failed

imyller, 2015-03-11:
There seems to be a lot of packages providing "virtual/...." so I assumed this is a safe naming convention. 

These are defined so that meta-nodejs and meta-iojs layers can be exchanged without updating depending recipes.

Would it be acceptable to change PROVIDES and RPROVIDES to "node" instead of "virtual/node"?

fladi, 2015-03-11:
Strange, during my build only nodejs turned up with this error. Using "node" as `(R)PROVIDES` is what I did now for my build. Of all the packages that were built until the point where it failed, only nodejs used the "virtual/" prefix. I'll see if any other package turns up with the same error in my current build.

imyller, 2015-03-11:
Difference might be that I'm using Ångstrom based distribution without .deb packaging and you are using Poky. Ångstrom layers contain lot of virtual prefixed (R)PROVIDES.

I'm going to commit a change to `PROVIDES = "node"`. I've already checked that it should not collide with commonly used OE packages.

If someone has added `PREFERRED_PROVIDER_virtual/node = "nodejs"` to their `local.conf` they might want to replace it with `PREFERRED_PROVIDER_node`

fladi, 2015-03-11:
You could have left `PROVIDES` with "virtual/node". It's only `RPROVIDES` that is used in the package generation.

imyller, 2015-03-11:
We're both too fast. Did not see your pull request coming ;) 

Will fix.

closed 2015-03-11 by imyller

fladi, 2015-03-11:
I should have sent my comment first and do the pull-request later :-)

Kudos for fixing it tough! Keep up the great work!

imyller, 2015-03-11:
Thank you. I'll fix the meta-iojs too (imyller/meta-iojs#1)

commit 0faf7195 (2015-03-11): fixed: (R)PROVIDES virtual/node conflicts with .deb control file format

Closes #3
commit d7152f68 (2015-03-11): Merge pull request #4 from fladi/fix-rprovides

Close #3: Use virtual package name "node" in RPROVIDES
commit ecd727e3 (2015-03-11): Close #3: Use virtual package name "node" in RPROVIDES

`RPROVIDES` specifies the provided virtual packages in the sense of
a package management system (ipk, deb or rpm) that a recipe provides.
Thus it does not need the "virtual/" prefix that is often used in
`PROVIDES`.