imyller/meta-nodejs #9: execvp:printf: Argument list too long

krunalsoni01, 2015-09-15:
Hi,
I have followed your steps for nodejs_4.0.0 but when I'm doing "bitbake node" it is is giving me the following error.
"Log data follows:
| DEBUG: Executing shell function do_compile
| make -C out BUILDTYPE=Release V=1
| make[1]: Entering directory `/yocto/pi2/poky/rpi-build/tmp/work/cortexa7hf-vfp-vfpv4-neon-poky-linux-gnueabi/nodejs/4.0.0-r1.0/node-v4.0.0/out'
| make[1]: execvp: printf: Argument list too long
| make[1]: *** [/yocto/pi2/poky/rpi-build/tmp/work/cortexa7hf-vfp-vfpv4-neon-poky-linux-gnueabi/nodejs/4.0.0-r1.0/node-v4.0.0/out/Release/obj.target/deps/openssl/libopenssl.a] Error 127
| make[1]: Leaving directory `/yocto/pi2/poky/rpi-build/tmp/work/cortexa7hf-vfp-vfpv4-neon-poky-linux-gnueabi/nodejs/4.0.0-r1.0/node-v4.0.0/out'
| make: *** [node] Error 2
| WARNING: exit code 2 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /yocto/pi2/poky/rpi-build/tmp/work/cortexa7hf-vfp-vfpv4-neon-poky-linux-gnueabi/nodejs/4.0.0-r1.0/temp/log.do_compile.30379)
"

Please help me out to solve this issue as soon as possible.
Thanks ,
Krunal

imyller, 2015-09-15:
Openssl bundled with Node.js source is sometimes causing compilation issues. This is the reason some earlier Node.js releases were actually not built with the bundled openssl and instead linked to shared libopenssl on the target system. Two were built separately and just had package dependency properly set. However, recent Node.js and io.js releases have been building just fine. This applies also to v4.0.0.

This error is more an issue of your build host system than the target platform, Node.js itself or the bitbake recipe used. I can confirm that Node.js 4.0.0 bitbake compilation works with x86_64 Ubuntu 14.04 LTS build host and Yocto 1.8 builds. On that platform, there are no issues with long argument lists. If you are using older Yocto (or Yocto based) releases, I can only suggest that you upgrade your build environment.

There is one easy thing you should check though: Yocto buildsystems often prefer that the default shell on the build host is set to `bash`. Ubuntu defaults to `dash` for example, which might cause issues with bitbake build scripts. Check your distribution specific instructions on how to change the default shell.

Other solution is to introduce a conditional which allows building Node.js recipe without bundled openssl and with libopenssl package dependency. I have not (yet) implemented that option into Nodej.s bb recipes.

krunalsoni01, 2015-09-18:
Hi Imyller,
 Thank you so much i really appreciate your quick response i have followed your comments and worked on it and it is working on my PC.Once again Thank you so much for helping me out. :+1:

imyller, 2015-09-18:
Glad to help. I'm closing this issue as resolved.

closed 2015-09-18 by imyller

mwarning, 2016-04-14:
I've still git this error on up to date Linux Mint and Ubuntu installs.
The solution was to move the build directory outside the home directory to /.
The margin of path length seem to be too narrow.

There seem to be some sort of patch that might be of help fixing this issue:
http://patchwork.openembedded.org/patch/72811/

imyller, 2016-04-14:
The patch linked applies to ancient Node.js 0.8.

I'd suggest you try compiling with recent meta-nodejs and Node.js 4.x or 5.x recipes.

mwarning, 2016-04-14:
@imyller I can compile your recipes without any patch or problems when I do it right on top of /. But it fails when I do it on top of my home directory.

I mentioned the patch because it might help. But I wasn't able to apply it.

imyller, 2016-04-14:
The real path length issue is with bundled OpenSSL.

You can opt to build Node.js recipe with shared OpenSSL using `PACKAGECONFIG` value `openssl`

http://www.yoctoproject.org/docs/2.0/ref-manual/ref-manual.html#var-PACKAGECONFIG

bachp, 2016-04-14:
Until now the only way to make this reliably work is by using external OpenSSL.

@imyller Maybe we should make the shared OpenSSL the default. The downside would be that who are using older OE versions that don't include OpenSSL 1.0.2 need to disable is explicitly.

armandciejak, 2016-04-20:
Hello,
I'm using meta-nodejs for compiling Node.js 4.x for a target that runs Yocto (1.5). I had the same issue than @mwarning, but I managed to apply the patch http://patchwork.openembedded.org/patch/72811/ by patching it. The result is available here https://github.com/riedonetworks/meta-nodejs on the dora branch.

@imyller if you think it is worth it I can prepare a pull request.

jsolla, 2016-07-19:
Any update on this issue? I tried to manually apply the patch provided by @armandciejak to Node 6.3.0 but not I get a "name too long" error.

bachp, 2016-07-20:
I can only recommend to use the OpenSSL bundled with OE instead of the internal one.

imyller, 2016-07-20:
I agree with @bachp 

Node.js internal OpenSSL has build issues with systems having limited path and/or command line length; especially when OE build path is already long and includes spaces.

I haven't yet found simple all-around solution to this and the same issue has been seen outside OE builds too for Node.

For my build servers to prevent path related build errors, I use shortest OE path possible (for example `/oe`) and avoid spaces in paths - and use OE openssl instead of Node.js internal.

mgiuncato, 2016-07-21:
@imyller : 
I fall in the same problem on ubuntu 14.04 , bitbaking for edison intel..

( sorry for my poor preparation on bitabke & C.)

how you do to ".. and use OE openssl instead of Node.js internal." ?
maybe you changed "nodejs_4.inc" in ; "

PACKAGECONFIG[zlib] = "--shared-zlib,,zlib"
PACKAGECONFIG[openssl] = "--shared-openssl,,openssl"

"
I'm very frustrated wth this error..
M.

imyller, 2016-07-22:
@mgiuncato 

Firstly, no - you do not have to modify any `meta-nodejs` internal files to make `PACKAGECONFIG` changes. `PACKAGECONFIG` is there to prevent that.

The error is caused by too long command line being run in Node.js internal OpenSSL compilation (several hundred source files with full path in one command). It is not easily fixable as it only pops up with some peoples build machines and is dependent on their folder structure, distro version and other settings.

At this time I do not have any more tips for you to try than:

* With Ubuntu, make sure your default shell is bash (as recommended by Yocto btw):
```sh
sudo update-alternatives --install /bin/sh sh /bin/bash 100
```

* Raise your program stack size with `ulimit`. For example:
```
sudo ulimit -s 65536
```
   or permanently with `/etc/security/limits.conf`. Search the web for docs about these if needed.

* Use as short as possible build directory path for your OE installation: 

My build servers have user `oe` with custom home directory set as `/oe` which also is the OE root (for example build dir is `/oe/build`). This matters for `argument list too long` errors for Yocto builds because many Makefiles link all sources with their full paths + all the compiler flags. For OpenSSL this file count is considerable and having a long OE build root path just makes the problem worse. This applies to few other OE package builds unrelated to Node.js too. So, anything like`/mnt/my_huge_drive/projects/yocto/poky/201607/test2/build` is sure to fail.

* Use Yocto `PACKAGECONFIG` system to configure Node.js recipe to always use shared OpenSSL and skip building the internal one (the one provided by OE core layer):

Docs: http://www.yoctoproject.org/docs/2.1/ref-manual/ref-manual.html#var-PACKAGECONFIG

For example `nodejs_%.bbappend` in one of your own layers with:

```bitbake
PACKAGECONFIG_append = " openssl"
```

or in your `local.conf` (or if you have custom distro, in your `distro.conf`):

```bitbake
 PACKAGECONFIG_append_pn-nodejs = " openssl"
```

jsolla, 2016-07-22:
@mgiuncato

My initial tests trying to setup the usage of openssl as a shared library failed while building node 6.3.0, looks like there's a different problem with this. For the time being I just used the --wihtout-inspector option as a workarond to build node wihtout openssl at all (found the solution somewhere on internet)

imyller, 2016-07-22:
I'm probably adding `v8-inspector` option as `PACKAGECONFIG` variable to next Node.js release.

For embedded systems allows minimizing newest Node.js versions a bit if one does not require the V8 inspector features.

I've found no problems when building Node.js without internal OpenSSL using `PACKAGECONFIG` with `openssl`. The recipe obeys that setting nicely and adds `--without-openssl` configure options etc. properly. It even automatically `RDEPENDS` on openssl so that you don't have to worry about that.

imyller, 2016-07-22:
@mgiuncato 

> looks like there's a different problem with this

Would you like to be more specific? Something that we can fix maybe?

jsolla, 2016-07-22:
@imyller 

Im now finishing some other stuff and will have to rebuild node 6.3.0 in 2 or 3 days, will try to build node with shared openssl library and try to clarify my affirmation of "different problem". Sorry for the lack of precission, just did a quick testing a few days ago and build was crashing whilst it was building fine just using the "--wihtout-inspector" option. I'll keep you posted on progress/errors.

mgiuncato, 2016-07-22:
@imyller : 
tks a lot, appreciate this fast response, and much more the description :-)

/bin/bahs: I use bash since times ago

change path: unfortunately bitbake hates the changes of name of directories requesting another global rebuild. this will take 6 hours on my i5 ubuntu 14.04 ... I hope this is the last resource.

sudo ulimit: this is simple and fast, I immediately do

@jsolla:
thanks but my node developers ask me ssl disponibility

imyller, 2016-07-22:
@mgiuncato Thanks. Waiting for your feedback.

Generally I'd like to add most common customizations as `PACKAGECONFIG` options to meta-nodejs recipes so that people wouldn't have to make custom versions of the main recipes for simple build configuration changes. V8 inspector is one of those that will be added soon.

imyller, 2016-07-22:
@mgiuncato 
>/bin/bahs: I use bash since times ago

Actually if you are using Ubuntu unmodified you are actually been using mostly bash compatible __dash__ ( from `/bin/dash`) as your default shell. Yocto recommends changing that to the real __bash__ for OE build machines.

imyller, 2016-07-22:
For all interested:

With todays release of Node.js v6.3.1 there is now new `PACKAGECONFIG` option for the recipe:

`v8-inspector`

By default V8 inspector is now __disabled__ for Node.js 6.x builds, but this `PACKAGECONFIG` can be added to enable it for your custom build.

As usual, use of shared OpenSSL (disables building of internal OpenSSL) can be requested with `openssl` `PACKAGECONFIG` option.

For more information about how to configure `PACKAGECONFIG` options see: http://www.yoctoproject.org/docs/2.1/ref-manual/ref-manual.html#var-PACKAGECONFIG

mgiuncato, 2016-07-22:
@imyller 
-  ulimit on my system says "unlimited"

i Will  try changing with shorter path...

stay tuned..

imyller, 2016-07-22:
Ubuntu 14.04 LTS unmodified install:

```shell
root@mgmt:~# ulimit -s
8192
```

You must have modified system default settings for program stack size to get `unlimited`. Usually the maximum value is `unlimited` though:

```shell
root@mgmt:~# ulimit -Hs
unlimited
```

mgiuncato, 2016-07-22:
@imyller 
verified /bin/bash not /bin/dash

imyller, 2016-07-22:
@mgiuncato Good, because default unmodified Ubuntu install is defaulting to `dash` for its system shell. Login shell is `bash` for the user, but that is irrelevant to the issue.

```shell
root@mgmt:~# readlink -f /bin/sh
/bin/dash
```

MartinPreinfalk, 2016-08-15:
Big THX for the information povided here! I ran into same issue today.

Host: Ubuntu 16.04 LTS
yocto: krogoth
shell: bash
path to build-dir: /home/martin/project/raspberryPi/jumpnowtek/rpi/build

adding:
`# avoid build internal openssl of node-js
PACKAGECONFIG_append_pn-nodejs = " openssl"
`
to local.conf did the trick

alvarez86, 2016-08-31:
I believe there is a way to solve this issue even with the bundled OpenSSL, at least on Linux.
What happens is the following:

1. The generated Makefile has a variable **builddir_name** which is then prepended to every object used on the command lines.
2. When nodejs tries to create the archive of OpenSSL, the prepended variable causes a really long line to be generated, and Linux prohibits this.
3. Thankfully, that variable is defined like this:`builddir_name ?=`, which means that anything you pass on the command line will overwrite the value of the variable.
4. So all you have to do is to add `builddir_name=./out` to the make command line, that will overwrite the value of the variable and you will get **./out/some_path/object.o** on your command line instead of **your_really_long_build_prefix_to_nodejs_build_dir/out/some_path/object.o**. The now much reduced command line should not give any problems, unless a really large number of objects is generated.

To test it you can just create a bbappend file with the following content:
`EXTRA_OEMAKE += "builddir_name=./out"`

I would be really glad if someone could test this solution and confirm it really works.

By the way, sorry for bad english or any mistakes, still learning my way around github.

imyller, 2016-08-31:
@alvarez86 Thank you for researching the root cause of this issue!

I think the solution to the problem is with modifying `builddir_name` as you suggested, but I'd still like to make sure that `./out` or whatever is still located in proper build dir provided by BitBake.

It's dangeous/hacky to build to unexpected location. It probably is not universal and won't work in all BitBake environments. 

Proper solution might be resolving short relative path (instead of absolute path) to build dir and providing that to `builddir_name`.

alvarez86, 2016-08-31:
@imyller, my solution does not work, it compiles, but then the installation fails, I still have to find out what's going wrong with it.

I will try to see if there's some way to modify only the openssl build, perhaps dealing with the **openssl.target.mk** file, but I can't promise anything.

alvarez86, 2016-08-31:
I will register it here just so people can try it, but as @imyller has said, it probably isn't the right solution,
be wary of using this, as it might not work on all environments, even then, I won't guarantee it will work
at runtime.

Just set builddir_name to **.**, as in `EXTRA_OEMAKE += "builddir_name=."`

Good luck to anyone that tries this, please let me know how well it works.

imyller, 2016-08-31:
I'll definitely experiment with this. If the build succeeds it should be quite trivial to get install task working.

lewicki-pk, 2016-11-02:
Hi. I have the same problem.
I use `nodejs_%.bbappend` file with

    PACKAGECONFIG_append = " zlib openssl"

in it.

I'm using latest mesta-nodejs and meta-nodejs-contrib and I'm trying to build `nodejs-6.9.1-r2.7`

It fails on do_compile task. Any Idea how to fix this?

imyller, 2016-11-02:
@lewicki-pk Did you resolve the issue already: https://github.com/imyller/meta-nodejs/issues/55#issuecomment-257858691 ?

lewicki-pk, 2016-11-02:
More of a workaround than fix, but yes.

If somebody gets here from google: change your yocto path to much shorter and this will do.

I hope this will be fixed in the newer versions of nodejs and also backported to v6..

harshlad07, 2022-09-07:
> PACKAGECONFIG_append_pn-nodejs = " openssl"

Not working