imyller/meta-nodejs #19: Add to OpenEmbedded layer index

bachp, 2016-01-08:
I would like to add this layer to the OpenEmbedded layer index: http://layers.openembedded.org/

But I would like to have the OK of the maintainer first.

scordibella, 2016-02-09:
Adding this layer to OpenEmbedded layer system will add a great value to both oe project and this layer.

imyller, 2016-02-09:
I agree, but as meta-nodejs maintainer would like to manage the OE Layer Index listing myself.

I've kept out of from the OE layer index mainly to let this layer stabilize and get to the level of quality I feel is worthy of such listing.

I waited also because the earlier separation of Node.js and io.js caused lot of trouble for packagers and future Node.js product naming was uncertain.

Now things have settled within Node.js project - and with meta-nodejs.

So list we will :+1

JubaHub, 2016-02-09:
I am happy to hear this from you @imyller  !
Thank you for your work!
I will wait the Yocto integration.

imyller, 2016-02-12:
OpenEmbedded Metadata Index is having some issues and layers can not be submitted until it is fixed.

meta-nodejs will be there as soon as possible.

--
[image]
--

imyller, 2016-02-13:
This has been reported to Yocto Project:

https://bugzilla.yoctoproject.org/show_bug.cgi?id=9106

imyller, 2016-02-19:
Layer submit bug has been resolved at Yocto Project.

meta-nodejs layer is now being reviewed for inclusion into OpenEmbedded Metadata Index

imyller, 2016-02-22:
meta-nodejs has been listed publicly:

http://layers.openembedded.org/layerindex/branch/master/layer/meta-nodejs/

closed 2016-02-22 by imyller