imyller/meta-nodejs #21: unable to install bcrypt, zmq, hiredis, qlobber, etc.

dlarm, 2016-01-14:
Compiling bcrypt node modules results in the following error:

g++: error: none: No such file or directory

reproduce by opening a bitbake devshell, and running npm install bcrypt

This seems to also affect several dependencies of bcrypt and others: zmq, hiredis, utf-8-validate, fs-ext, bufferutil

bachp, 2016-01-14:
Thanks for reporting the issue, But could you provide more details please. For example provide the recipe that is failing.

dlarm, 2016-01-14:
Thanks for the follow up, I just created a simple recipe to isolate the issue, and it appears the compile fail was related to use of allarch, and not related to your npm layer options. Thanks for making this layer available!

closed 2016-01-14 by dlarm