imyller/meta-nodejs #22: Unable to install versions other than 0.8.18

tremendus, 2016-03-01:
It's probably me ... sorta new to Yocto ... but maybe not. I have installed the layer per readme.md installation, steps 1 and 2. 

```
# sources/layers.text
meta-nodejs,git://github.com/imyller/meta-nodejs.git,master,HEAD
```

I have added the following to my local.conf: 

```
# 
# NodeJS
# 
PREFERRED_VERSION_nodejs = "4.%"
DEPENDS += " nodejs"
RDEPENDS_${PN} += " nodejs"
```

I then try to bitbake and I can bake node .... but only 0.8.18, with the following message in console

```
Parsing of 1515 .bb files complete (0 cached, 1515 parsed). 1945 targets, 163 skipped, 6 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
NOTE: preferred version 4.% of nodejs not available (for item nodejs)
NOTE: versions of nodejs available: 0.8.18
NOTE: preferred version 4.% of nodejs not available (for item nodejs-dev)
NOTE: versions of nodejs available: 0.8.18
```

I have also tried specifying exact versions 4.3.1 and non-4.x versions but all throw the same note in console. 

Bug? Am I missing something? Is this a bug?

tremendus, 2016-03-01:
Never mind, seems like node recipe is being provide by other layers, and my nodejs layer isn't available as a provider or preferred provider, must be on my end .... closing

closed 2016-03-01 by tremendus

imyller, 2016-03-02:
Good to hear that the issue resolved itself.

I've never released recipe for0.8.18 in meta-nodejs so it's obvious that it was coming from some other recipe layer.

bachp, 2016-03-02:
Just a side note. It seems like [meta-openembedded](https://github.com/openembedded/meta-openembedded/commit/80fec0a9842c12dc6ba39d4383d0ac4e66e60037) now contains nodejs 4.3.1 too.