imyller/meta-nodejs #23: discussion: deprecation and removal of io.js recipes

imyller, 2016-03-02:
I'm currently considering removing io.js recipes from meta-nodejs.

Io.js is no longer receiving upstream updates and is replaceable with Node.js 4.x LTS for most users. 

If io.js recipes are removed from this layer, I'll probably add an package alias `iojs` to Node.js LTS recipe so that possible dependencies to `iojs` package will function.

imyller, 2016-03-02:
Request for comments.

/cc @bachp @jmesmon @geomatsi @gsikorski @fladi

bachp, 2016-03-02:
I also think for iojs users and update to Node.js 4.x is feasible.

jmesmon, 2016-03-02:
My use case was never updated to `iojs`, so I don't have any issue with this.

geomatsi, 2016-03-02:
My use case is trivial: I only experimented with embedded nodejs builds and didn't use iojs at all. In general, IIUC iojs is going to be discontinued. So removing those recipes will help to make the right choice for new users.

imyller, 2016-03-02:
Thank you for your feedback. Implemented this change in PR #25

closed 2016-03-02 by imyller