imyller/meta-nodejs #24: discussion: follow only semver-major Node.js releases

imyller, 2016-03-02:
Node.js uses semantic versioning and currently meta-nodejs layer maintains recipe for every semver-minor release. This has proven to be problematic.

About semantic versioning @ http://semver.org
```
Given a version number MAJOR.MINOR.PATCH, increment the:

MAJOR version when you make incompatible API changes,
MINOR version when you add functionality in a backwards-compatible manner, and
PATCH version when you make backwards-compatible bug fixes.
Additional labels for pre-release and build metadata are available as extensions to the MAJOR.MINOR.PATCH format.
```

Recently Node.js has released semver-minor updates to LTS moving from 4.2 to 4.3 and soon 4.4. This means that LTS actually is any 4.x Additionally all previous stable semver-minor releases in 5.x line are left unmaintained urging users to keep up with the latest semver-minor release.

It would be more proper to maintain OpenEmbedded recipes for each server-major (0.10, 0.12, 4.x, 5.x) release rather than separate one for every semver-minor release.

If there are no objections, I'll remove unmaintained semver-minor releases and continue releasing separate recipes for only semver-major release lines.

imyller, 2016-03-02:
Request for comments.

/cc @bachp @jmesmon @geomatsi @gsikorski @fladi

gsikorski, 2016-03-02:
OK. We use a clone of this repo anyway and it is pretty easy to change minor version.

bachp, 2016-03-02:
For me it would be sufficient to have the LTS (4.x) and Current version (5.x) supported.
So the change you propose is perfectly fine with me.

jmesmon, 2016-03-02:
Sounds fine to me.

geomatsi, 2016-03-02:
Fine by me.

imyller, 2016-03-02:
Thank you for your feedback. Implemented this change in PR #26

closed 2016-03-02 by imyller