imyller/meta-nodejs #28: How to use this layer for native only

mschuckmann, 2016-03-22:
I may be missing something here, I want to use this layer native only. 
I have a recipe for a web-app that needs to get built using npm, bower, and grunt and the result installed on the image. The image doesn't need npm,bower or grunt only the build system. 
Looking at the npm class node always gets added to RDPENDS.

imyller, 2016-03-25:
You can put `DEPENDS` on `nodejs-native` to get native (but "cross-compiling") Node.js and npm. `npm` and `npm-install` bitbake classes included in this layer automatically depend on `nodejs-native` and provide some helper commands and build tasks.

You do not need to create `RDEPENDS` run-time dependencies to any of the packages.

I personally have a use-case where web-app is built the way you described and results are packaged for target system.

imyller, 2016-03-25:
> Looking at the npm class node always gets added to RDPENDS.

That is true. As soon as I have some time, I'll look in to providing build system only helper classes that are more focused on running build automation tools.

imyller, 2016-03-29:
This should be fixed now with commit 247732c8fcf80c96c8e403204005f384186e3738

There is no longer forced RDEPENDS to Node.js package when using `npm` class.

closed 2016-03-29 by imyller