imyller/meta-nodejs #31: reason for `--without-snapshot ` in configure

mbroadst, 2016-04-11:
Hi, I'm trying to cross-compile rethinkdb using yocto which includes a static build of v8, and I've been running into problems with the generated snapshot utility. I noticed in your recipes that you disable the snapshot, is that because you couldn't get it working? Just wondering if you could help with some insight here, thanks!

imyller, 2016-04-13:
Yes, the generated `mksnapshot` also gets cross-compiled and you'd need native version of it for v8 snapshot build to succeed.

Here is more info about current situation with v8 snapshots and Node.js cross-compiling:
https://github.com/nodejs/node/issues/4860

I have an option to compile `nodejs-native` as a dependency before `nodejs` recipe, and then copying over the `mksnapshot` natively compiled for the build host. This approach should work but it's bit ugly and requires "unnecessary" compilation of the native package every time target package gets updated.

There is some discussion about exporting `CC_host` and `CXX_host` environment variables before building allowing some kind of partial cross-compilation, but I havent tried this approach yet with Yocto packages. (https://github.com/nodejs/build/issues/266#issuecomment-171660151)

mbroadst, 2016-04-14:
@imyller hmm interesting, so potentially we could just modify the gyp file for building `mksnapshot` to override the compiler used. I'll give it a shot later today and report back, thanks

mbroadst, 2016-04-15:
@imyller hey, so I did some quick read throughs of the build system and tried this out, which seems to work:

```
do_configure () {
  export LD="${CXX}"
  alias g++="${CXX}"
  GYP_DEFINES="${GYP_DEFINES}" export GYP_DEFINES
  ./configure   --prefix="${prefix}" \
                --dest-cpu="${@map_dest_cpu(d.getVar('TARGET_ARCH', True), d)}" \
                --dest-os=linux ${ARCHFLAGS} \
                ${EXTRA_OECONF}
  unalias g++

  # testing with-snapshot
  sed -i "s#'want_separate_host_toolset%': 0#'want_separate_host_toolset%': 1#g" common.gypi
  sed -i "s#'want_separate_host_toolset': 0#'want_separate_host_toolset': 1#g" config.gypi
}
```

specifically I modified the `do_configure` to enable snapshots (the default, so just remove `--without-snapshot`), and then I explicitly defined `want_separate_host_toolset` to 1 with sed. I couldn't figure out how to define it through the configure script maybe you have some insight there.

So it builds fine for the native and host packages at least for x64 <=> qemux86_64. There are a number of concerns/questions/thoughts I have at this point:

  * I think this is probably needlessly creating two toolchains in _at very least_ the native package, so it should probably be removed for the native's `do_configure`

  * As I mentioned above sed probably isn't the ideal solution here

  * I think this flag enables more work than just what's needed for `mksnapshot` though I'm not quite sure _what_ exactly - as I'm just passing through quickly here

  * I have done absolutely no testing on anything other than my current environment, which doesn't even target real hardware presently (currently just building a VM image during early development)

HTH

imyller, 2016-04-20:
Great work!

I see that there is some logic for defining `want_separate_host_toolset` in Node.js `configure`:

https://github.com/nodejs/node/blob/v5.10.1/configure#L694

We might be able to avoid using sed by specifically defining `host_arch` and `target_arch` for `configure`?

mbroadst, 2016-04-20:
I tried that approach, and I'm sure I missed something, but this value was set explicitly no matter what I did

imyller, 2016-04-21:
ref #34

imyller, 2016-10-29:
V8 snaphots are being worked on in `master` branch for Node.js v7 and already works with certain limitations.

There are unresolved issues with cross-compilation dependencies affecting `mksnapshot` and `mkpeephole` build steps - discussed in #56

closed 2016-11-02 by imyller

markrey, 2017-01-31:
@mbroadst Is it possible to share your Yocto Layer file for rethinkdb? (if not confidential)
I am struggling to do since last few days :)

markrey, 2017-02-02:
@mbroadst Thank you very much! Really do appreciate it! Long night tonight to compile this :)

markrey, 2017-02-02:
@mbroadst one file 0004-remove-no-registry-flag-for-npm-install.patch is missing. Do you have it available by any chance?

mbroadst, 2017-02-03:
@markrey oops  yeah here it is: https://gist.github.com/mbroadst/b4f4b11d22e8fd9529b4f01507f31845