imyller/meta-nodejs #32: Add tags for versioned releases

olafmandel, 2016-04-15:
When working with a specific commit of the repository (e.g. v0.12.13), specifying that by a tag-name instead of commit hash (e.g. a94d31d) for the `git checkout` command would be much easier. And when cloning, it is possible to directly go to specific tag but not to an untagged commit, e.g. `git clone --branch v0.12.13 git://github.com/imyller/meta-nodejs.git`.

One can generate a list of commands with:

    git log --oneline | \
    sed -n 's/\([[:xdigit:]]\+\) .*\(\([[:digit:]]\+\.\)\{2\}[[:digit:]]\+\).*/\2 \1/p' | \
    grep -v -e 4127cbc$ -e 64a33f6$ | \
    sort -V | \
    sed -e '/^.\{13\}$/s/[^ ]*$/ &/' -e '/^.\{14\}$/s/[^ ]*$/ &/' -e 's/^/git tag v/'

Currently, this generates these commands:

    git tag v0.8.28  7abfef5
    git tag v0.10.26 75aeeee
    git tag v0.10.27 3da27dc
    git tag v0.10.28 5b960f6
    git tag v0.10.30 261eee9
    git tag v0.10.31 1b49519
    git tag v0.10.32 1797d84
    git tag v0.10.33 20f9a28
    git tag v0.10.34 ac168aa
    git tag v0.10.35 14617df
    git tag v0.10.36 a447eac
    git tag v0.10.37 bca2997
    git tag v0.10.38 bdb1a14
    git tag v0.10.39 af20168
    git tag v0.10.40 bf4d4bb
    git tag v0.10.41 836a8cd
    git tag v0.10.42 fefa471
    git tag v0.10.43 c92d743
    git tag v0.10.44 db8609a
    git tag v0.12.0  569079f
    git tag v0.12.1  eb7d94d
    git tag v0.12.2  bc03adb
    git tag v0.12.4  6eb2941
    git tag v0.12.5  c1a791d
    git tag v0.12.6  396f88b
    git tag v0.12.7  c70b4a0
    git tag v0.12.8  b6d2f13
    git tag v0.12.9  3901c5e
    git tag v0.12.10 f63207b
    git tag v0.12.11 9e73e76
    git tag v0.12.12 8e0bcba
    git tag v0.12.13 a94d31d
    git tag v1.1.0   d70cc3e
    git tag v1.3.0   0889b78
    git tag v1.4.1   ab612b0
    git tag v1.4.3   dda7c4c
    git tag v1.5.0   8050bfc
    git tag v1.5.1   b6a1f2c
    git tag v1.6.1   d204919
    git tag v1.6.2   16a15d6
    git tag v1.6.3   c480150
    git tag v1.6.4   5c03252
    git tag v1.7.1   b589db2
    git tag v1.7.7   08d53c4
    git tag v1.8.1   f5eb821
    git tag v1.8.2   3d86cdf
    git tag v1.8.3   d48a1cc
    git tag v2.0.0   4870970
    git tag v2.0.1   f6a853a
    git tag v2.0.2   7734fec
    git tag v2.1.0   6b95527
    git tag v2.2.1   83e1854
    git tag v2.3.0   dd6803f
    git tag v2.3.1   883a96f
    git tag v2.3.3   9417bb9
    git tag v2.3.4   65d629f
    git tag v2.4.0   a7da7d3
    git tag v2.5.0   facad22
    git tag v2.9.1   bdda4aa
    git tag v3.0.0   01d1cd3
    git tag v3.1.0   c618c64
    git tag v3.2.0   d3264da
    git tag v3.3.0   82186b6
    git tag v3.3.1   86ec95b
    git tag v3.9.1   a5e3433
    git tag v4.0.0   c85673e
    git tag v4.1.0   28b35aa
    git tag v4.1.1   fa5cfb1
    git tag v4.1.2   38c8673
    git tag v4.2.0   8d965dc
    git tag v4.2.1   05a64f3
    git tag v4.2.2   626d280
    git tag v4.2.3   cfebec3
    git tag v4.2.4   95461cb
    git tag v4.2.5   204a14f
    git tag v4.2.6   fbedd13
    git tag v4.3.0   ea69835
    git tag v4.3.1   4812706
    git tag v4.3.2   26e0b18
    git tag v4.4.0   1d521a2
    git tag v4.4.1   ac01d7d
    git tag v4.4.2   17c1112
    git tag v4.4.3   c934588
    git tag v5.0.0   c372f0c
    git tag v5.1.0   2db2f32
    git tag v5.2.0   71b1316
    git tag v5.3.0   8aedc96
    git tag v5.4.1   4d78558
    git tag v5.5.0   9cb8659
    git tag v5.6.0   af45880
    git tag v5.7.0   9438cb3
    git tag v5.7.1   555fc5f
    git tag v5.8.0   dc87662
    git tag v5.9.0   4a6cc57
    git tag v5.9.1   1d97c89
    git tag v5.10.0  59af074
    git tag v5.10.1  634d88e

Yes, 97 tags for 239 commits is a very high percentage, but that is how this repository works...

imyller, 2016-04-15:
The problem with tagging Node.js releases in this repo is that this repo also keeps compatibility with OpenEmbedded Yocto releases. For example, once Yocto updates, the recipe for v5.10.1 Node.js could change too and git tag would be left pointing to incompatible version. For example v0.10 or v0.12 recipes are no longer the same they were first released.

So, recipe originally written for specific Node.js version might change multiple times as new Yocto releases come; corresponding versioned tag should also be kept up-to-date. 

Tagging only the first commit for specific Node.js release is dangerous and possibly quite misleading for users.

If you have good solution to this, I'm always open for suggestions.

olafmandel, 2016-04-15:
> For example v0.10 or v0.12 recipes are no longer the same they were first released.

I was confused here for a second: the actual receipe files for e.g. 0.10.44 or 0.12.13 have not changed. But the used classes _did_ change, so I now see your point now.

> So, recipe originally written for specific Node.js version might change multiple times as new Yocto releases come; corresponding versioned tag should also be kept up-to-date. 

There are fewer Yocto releases than nodejs releases, so how often is that a problem?

I think tagging a specific version (three-tuple) should be fine because these basically claim that a specific file like e.g. `nodejs_0.10.44.bb` is available. As these get deleted once the next version comes out (e.g. db8609ae deleted `nodejs_0.10.43.bb`), there is relatively little change over the availability of that file. And if someone is really interested in having a specific (old) version of nodejs together with a bleeding-edge Yocto, they probably have to patch the repository anyway: they would need to work in master, but checkout the old recipe file.

I can't say about other people, but I could live well with being locked to an old Yocto-version if I insist on an old nodejs version...

imyller, 2016-04-20:
Currently meta-nodejs keeps only the most recent version of each release line supported by Node.js project:

* `5.x.x` (Stable)
* `4.x.x` (LTS)
* `0.12.x` (LTS)
* `0.10.x` (Maintenance)
* `0.8.x` (Legacy)

Within release old version is always replaced/removed by the new. My suggestion is that people define their dependencies to specific *release line* rather than specific version. That fits well with the compatibility promise Node.js makes in their semver-versioning scheme.

In this repo, the more correct approach would be to provide separate branches for each Yocto release (fido etc.) and maintain compatible versions of major Node.js release line within those. 

I personally have ability to reliably maintain only for one Yocto release (latest stable), so I haven't ventured there yet.

So, tagging specific Node.js releases is not the way I'd do it but I'm willing to consider maintaining each release line in Yocto release branches. This is the way most Yocto layers do it - including core layers.

Any thoughts on this @olafmandel ?

olafmandel, 2016-04-20:
> My suggestion is that people define their dependencies to specific release line rather than specific version. That fits well with the compatibility promise Node.js makes in their semver-versioning scheme.

That is true as far as nodejs or any of the recepies are concerned. It is not true for compatibility of the classes provided by the layer (so if another layer uses those classes) and possibly also not true for backwards-compatibility to older Yocto releases.

I must admit to not having debugged my problems further when I ran into build-problems after a `git pull` but to simply using the old commit again. This is certainly not ideal in the sense of me using old software (decoupled from security fixes).

> In this repo, the more correct approach would be to provide separate branches for each Yocto release (fido etc.) and maintain compatible versions of major Node.js release line within those.

For supporting users with different Yocto versions that always want the newest version: correct. But I came to this issue/request because I was in the process of writing the manual for how to recompile a firmware image and that manual starts like this: "prepare the build-environment like so:"

    mkdir -p <dir>
    git -C <dir> clone --branch <tag0> <yocto-url>
    git -C <dir> clone --branch <tag1> <layer1-url>
    git -C <dir> clone --branch <tag2> <layer2-url>

Here I want _precisely_ the same software versions, even if they are old. Currently, the line for meta-nodejs reads like so:

    git -C <dir> clone git://github.com/imyller/meta-nodejs.git
    git -C <dir>/meta-nodejs checkout <tree-ish>

So (for me) this is less about supporting the newest nodejs version on either the newest or an older Yocto release but it is about recompiling precisely the old firmware.

And yes: it is very difficult to get many of the npm commands to work without network interaction...

imyller, 2016-04-20:
Thank you for your case example @olafmandel 

I still do not see meta-nodejs tags as a straight solution to your case. If we use specific node versioned tags in this repo and introduce yocto release based branches, the tag could end up pointing to a different place at later time and cause you unexpected change very difficult to detect - making things worse.

What most Yocto distributions do is that they define their dependencies via specific Yocto release branch of the layer repo and then either to `HEAD` of that branch or specific commit hash. Problem with tags is that they could point to different release branch and/or change the underlying hash.

My use case example of defining firmware repos using the `repo` tool:

https://github.com/Ambertec/aeos-manifest/blob/jethro/default.xml

If I want to lock specific firmware down, I'd keep the layer repos pointing to the correct Yocto release branch but set locked commit hashes. I wouldn't trust tags, because repos could update those.

olafmandel, 2016-04-20:
Thank you for showing an example of the `repo` tool. I may consider its use at a later date (don't want to add another dependency right now) and for the helpful discussion.

I will close the issue now as tags for specific nodejs versions seem to be a bad fit for this layers development/release model.

closed 2016-04-20 by olafmandel