imyller/meta-nodejs #33: TARGET_ARCH and ARCH should use `map_dest_cpu`

mbroadst, 2016-04-17:
I've run into an issue where `node-ffi` fails to build because it depends on correct arch flags being passed to node-gyp. In this case the following variables should (I believe) not pass yocto's arch strings, but rather convert them to the architectures node cares about:

* https://github.com/imyller/meta-nodejs/blob/master/classes/npm.bbclass#L11
* https://github.com/imyller/meta-nodejs/blob/master/classes/npm.bbclass#L37

and it looks like you've already implemented this [here](https://github.com/imyller/meta-nodejs/blob/master/recipes/nodejs/nodejs_4.inc#L19) for the actual node build.

Does this seem incorrect to you? Otherwise I can put together a PR for this.

imyller, 2016-04-20:
That is absolutely true. I'd be happy to receive a PR if you're willing to implement this.

imyller, 2016-05-06:
Implemented

closed 2016-05-06 by imyller