imyller/meta-nodejs #38: Request for comments: Node.js helper classes/recipes layer

imyller, 2016-04-28:
In the recent months `meta-nodejs` has received number of contributions in form of helper classes/recipes (grunt, bower, etc..). More or less these are sidesteps from primary goal: maintaining a rock-solid Node.js runtime recipe layer.

In interest of keeping things responsibly maintained I've considered starting a sibling layer `meta-nodejs-contrib` (or similar) to separate core Node.js runtime recipes from various helper classes and Node-based software recipes.

For that project, I'm looking for interested parties as contributors/maintainers for keeping the included recipes up-to-date both with upstream software releases and Yocto-releases. 

I'd appreciate any feedback.

imyller, 2016-04-28:
/ccing some recent recipe/class contributors:

@bachp for Gulp and Grunt
@posborne for Gulp
@geomatsi for Bower
@mdavis777 for Typings

bachp, 2016-04-29:
We decided to go a different route than originally intended and we are no longer actively using the gulp and grunt classes. So I think it's better maintained by somebody else actually using it.

As for splitting these into another layer: A long time ago I had the idea to create a `meta-javascript` layer that provides recipes and classes for web developers. But I didn't pursue this to the end because the use case was no longer there for me. But if enough people show interest it might be useful. Maybe this is something to ask on the OE-devel mailinglis?

posborne, 2016-04-30:
I cannot claim to be a gulp expert, but I am currently using it for a project and would be willing to help with maintenance.  I'm not sure splitting makes sense -- there isn't too much harm in having a few extra recipes (and classes) available as these are still opt-in (parsing adds minimal overhead if they are not used).

As with the meta-rust (https://github.com/jmesmon/meta-rust) layer, I think there is a  bit of an unsolved question of how much should of an applications dependencies should be modeled within Yocto.  If we take the position of only providing the runtime and utilities for building applications based on npm/gulp/grunt/etc., that minimizes the number of required recipes.

imyller, 2016-05-06:
How about:

a) moving gulp, grunt, bower etc. to `recipes-contrib` dir and providing separate `README-contrib.md` for those? README.md is already getting too long and confusing.

b) New repository and layer `meta-nodejs-contrib` and moving relevant recipes and classes there. Proper README's for each helper class.

c) ?

Partially related to this: I'm soon renaming `recipes` to `recipes-devtools` to better match Yocto core directory layout.

bachp, 2016-05-07:
A middle way between a) and b) would be to have both the contrib and the current nodejs in the same repository. But still make them separate layers. This would be similar to how meta-openembedded works.

imyller, 2016-06-01:
@bachp @posborne 

I tried organizing the middle way you proposed, but encountered an issue with Bitbake with following folder structure:

```
meta-nodejs ->
  classes
  conf
  meta-nodejs-contrib ->
    classes
    conf
    recipes
  recipes-devtools
```

The second layer placed directly under other layer causes warnings/errors with `BBFILE_PATTERN` and `BBPATH` resolution because the parent path for `meta-nodes-contrib` layer is already included as a layer.

```
WARNING: No bb files matched BBFILE_PATTERN_nodejs-contrib '^/oe/aeos/layers/meta-nodejs/meta-nodejs-contrib/'
```

This does not happen with `meta-openembedded` because the repository root itself is not a layer root.

Resolution would be to move everything in this repo to a subfolders each hosting a separate layer, but that would cause too much breakage for existing users.

Only clean solution I can think of is to split the `meta-nodejs-contrib` to its own repository and keeping a configured layer dependency to `meta-nodejs` layer.

Your thoughts?

bachp, 2016-06-02:
You are right placing a layer inside another doesn't work.
So if you don't want to move the current layer, I think the only solution is to move it to a separate repo as you proposed.

imyller, 2016-06-02:
I think that is a proper way to do it. I'll make sure to document the existence of contrib repo well in the main repo.

I like to keep the main `meta-nodejs` stable and well aligned with Node.js upstream goals.

Contrib repo does not have to be so strict and I hope it attracts more outside collaboration and input.

@bachp still interested in maintainer status?

bachp, 2016-06-02:
@imyller Yes I would still be happy to help out.

imyller, 2016-06-16:
I'm happy to announce that a layer has been released:

https://github.com/imyller/meta-nodejs-contrib

It is in process of being reviewed and listed to OpenEmbedded Layer Index.

closed 2016-07-08 by imyller