imyller/meta-nodejs #39: Not compatible with Yocto 2.1

Aaronov, 2016-05-05:
There is a class name clash when using latest Yocto 2.1 which provides an "npm" class. May need to rename meta-nodejs class to something else?

imyller, 2016-05-06:
This is a huge issue right now.

Yocto's `npm` class does not offer similar functionality. `meta-nodejs` `npm` class focuses on calling the `npm` in Yocto way and offers `oe_runnpm` command allowing recipes to define their own install or compilation tasks while having reliable way to call cross-compiling `npm`.

Yocto 2.1 `npm` class matches `meta-nodejs` `npm-install` class in purpose, but fails to provide registry URL, global install, dedupe or http(s) proxy handling. Also it's not fully cross-compilation aware when compiling dependencies using V8 native API as there is no arch identifier mapping (differs for node-gyp and Yocto)

http://cgit.openembedded.org/cgit.cgi/openembedded-core/tree/meta/classes/npm.bbclass?h=krogoth

As Yocto has already done its 2.1 release there are very few options. Name clash is unfortunate and `meta-nodejs` likely has to rename its core class. This will cause a lot of breakage for Node.js recipe maintainers but it seems to be only way to resolve the conflict at this point.

As a side note, `meta-nodejs` will soon branch into master, jethro and krogoth branches - which will help to contain the situation a bit.

Good ideas are welcome.

/cc @otavio @posborne @arfoll

Aaronov, 2016-05-06:
Agree that rename is likely the only way at this stage. For now I simply deleted the Yocto npm.bbclass so I could continue to use meta-nodejs (which I prefer). Does the Yocto npm support use of package.json? This is what I use with meta-nodejs and I really like that.

imyller, 2016-05-06:
Yocto 2.1 `npm` does not use the npm's own fetcher, which causes lot of compatibility issues and native API compatibility issues. 

I'm currently discussing with Yocto folks about how to best resolve this.

arfoll, 2016-05-10:
When I wrote the npm fetcher and class in Yocto my main concern & need was to make sure everytime you made a build of a node pkg you where sure to get the exact same stuff. I added support for shrinkwrap so that you could provide a file that would be taken into account to make sure versions where exact, but shrinkwrap is not enough because it does not guarantee that versions don't change what they mean upstream, and because resolution of the dep is only done later the content is not kept in downloads/ nor is it available via the archiver, and we where concerned about people shipping node.js packages where they actually weren't sure of whether they'd be able to rebuild the same thing or even what licenses their resulting package included. That's why I downloaded everything to npmpkg and added support for lockdown.

You mention compatibility and native API issues, do you have any test cases that fail?

In your email you mention that Yocto's npm class expects stuff to be in a particular downloaddir, is that something you feel should be int he metadata just like when we have `S = "${WORKDIR}/git"` for git packages?

bachp, 2016-05-10:
@arfoll I think the discussion at https://github.com/npm/npm/issues/4210 might be of interest as this would allow to use the npm fetcher and then just store the resulting bundle.

imyller, 2016-05-11:
My toughts: 

With `meta-nodejs` and `npm` + `npm-install` classes I chose to offer native Node.js and npm experience rather than modify/replace npm's internal dependency resolving, deduping, shrinkwrapping etc. I still believe this is the proper way to implement it. npm is evolving project and we are already shipping multiple npm major versions with Node.js v0.8, v0.10, v0.12, v4, v5 and v6 recipes; each with different functionality. Not by fault but by design.

Recipes using `meta-nodejs` `npm` and `npm-install` classes will be able to bundle their dependencies according to `package.json` and build appropriately cross-compiled packages (ipk,deb, etc.) which are not going to change afterwards. Also, many Node.js applications are not just npm installing `node_modules` but running build tools, minifiers, compilers controlled by npm install process. We support all those because we leave it to native npm and developer. Further, if `package.json` is shrinkwrapped later builds will include the exact same versions with same licensing; as promised by npm.

I do understand the design rationale behind Yocto 2.1 `npm` class and bitbake fetcher, but in many ways it collides not only in class naming, but also in offered flexibility and compatibility with npm ecosystem.

I see little possibilities at this point to convert Yocto's implementation to match ours. It is not going to happen. On the other hand, I'd hate to lose all the features and flexibility of `meta-nodejs` implementation, which I believe also to be the correct 100% Node.js compatible way of doing it.

So, rename we will. `npm.bbclass` has to go away. 

For reference `oe_runmake` is defined in `base.bbclass`, so there is nothing to borrow from there. `oe_runnpm` could be defined in `nodejs.bbclass` or `runnpm.bbclass` or .. suggestions welcome!

I wouldn't change `npm-install.bbclass` because many recipes are depending only that and not `oe_runnpm` providing `npm.bbclass` directly. Those that have custom build process using `oe_runnpm` will have to migrate.

imyller, 2016-05-11:
@arfoll 
> You  mention compatibility and native API issues, do you have any test cases that fail?

You have to provide `--arch` and `--target-arch` (for gyp) and proper `CCACHE` and `LD` env variables before any call to npm for it to be fully cross-compiling capable. Do not forget that `install` is not the only command that might require gyp. Some packages have defined custom commands that have to be fully supported. That is exactly what our `oe_runnpm` does; in analogy what `oe_runmake` does for `make`. 

As an example, I expect `node-serialport` as package.json dependency (direct or sub-dependency) to fail cross-compiling with Yocto 2.1 `npm.bbclass` because it does not prepare the cross-compiling build environment for npm. Also packages that do recursive calls to `npm` in their install process will fail because of `no-registry` setting.

arfoll, 2016-05-11:
@imyller saying that package.json will include same version and licensing is incorrect, an npm package can be uploaded to npmjs witht he same version replacing a previous version. Obviously there's also the security risks of not checking any dep md5 against a known manifest etc... License checking is simply not taken into account, with your design it's not possible to md5 check licenses in a dependency of a package or have them be spit out in your image manifest.

I don't understand why you remain convinced only your design can cross compile/minify etc... We still call `npm install` so all scripts will get called just the same just in the `do_compile` instead of the `do_fetch` stage. I'm sure that'll break the odd thing but there's plenty of broken packages on cross compilation too.

We probably need to add `--arch` and `--target-arch` then, I'll check with node-serialport as an example. Stuff that calls `npm` recursively is interesting, do you have ane example of that?

imyller, 2016-05-11:
@arfoll 
Consider the following snippet from bb recipe:

```bitbake
inherit npm

do_compile() {

    # Compile a production build

    cd ${S}
    oe_runmake clean # Clean deploy dir
    oe_runnpm install --production=false  # Populate node_modules
    oe_runnpm do-deploy # Run do-deploy tasks in package.json to package Node.js app in single js  
    oe_runmake all # Compile deploy dir

}
```

My point is: Just like `oe_runmake` is important tool in Yocto. `oe_runnpm` is too.

New Yocto `npm.bbclass` provides `npm install` as you said, but it does not provide the full power of npm as a flexible build tool that it is.

Just running `npm install` is not enough. It is making too much assumptions about Node.js application described in the recipe. Also, not everything is a Node.js application. NPM is the package manager for many other ecosystems too. `npm.bbclass` in Yocto make too many assumptions for the user.

I do get that there is remote possibility that someone uploads same version with different licensing to npmjs. That causes license conflict. I'd handle that in my recipes with Yocto's own `LIC_FILES_CHECKSUM` which allowing recipe maintainer to verify licensing.

imyller, 2016-05-11:
I correct myself:
> I'd handle that in my recipes with Yocto's own LIC_FILES_CHECKSUM which allows recipe maintainer to verify licensing.

You are correct. This does not work. Locked licensing remains an issue with pure npm solution.

imyller, 2016-05-12:
@arfoll
> Stuff that calls npm recursively is interesting, do you have ane example of that?

Consider following `package.json` snippet from real project:

```json
  "scripts": {
    "build": "node build.js",
    "serve": "node server.js",
    "gzip": "find build -type f \\( -name '*.html' -o -name '*.css' -o -name '*.js' -o -name '*.xml' -o -name '*.json' \\) -exec gzip -kf9 {} \\;",
    "deploy": "npm run build && npm run gzip",
    "load-versions": "node scripts/load-versions.js",
    "start": "npm run serve",
    "test": "npm run test:lint && npm run test:unit && npm run test:smoke",
    "test:lint": "standard && htmllint **/*.hbs && stylint layouts/css",
    "test:unit": "tape tests/**/*.test.js | faucet",
    "test:smoke": "tape tests/*.smoketest.js | faucet"
  }
```

Not every project is buildable or deployable with simple `npm install`.

bachp, 2016-05-12:
The fact that the `npm.bbclass` of `meta-nodjs` is bypassing the yocto download and mirror system caused some problems for us. We worked around this with an internal npm registry.
This is fine as this is the way how `meta-nodejs` was working. Not duplicating all the npm fetching logic avoided a lot of complexity and kept the layer very simple and clean.

However if Node.js is part of `meta-oe` or even `oe-core` I think it is mandatory to adhere to the conventions there and use the proper bitbake fetcher infrastructure. However when doing this we should try to leverage as much from npm as possible. Maybe this even means improving npm upstream.

@arfoll BTW similar issues should arise for other package managers lik PyPi (Python), Gems (Ruby) etc. How is it solved there?

imyller, 2016-05-12:
@bachp Internal npm registry is a good way to solve that.

Pressing issue is the class name collision with Yocto 2.1. That has to be resolved by renaming.

Yes, Node.js is distributed also by OE/Yocto, but there it severely lacks behind Node.js projects own release cycle. This is one of the biggest reasons for `meta-nodejs` exists; in addition to helper classes provided by `meta-nodejs` allowing access to `npm` ecosystem and more feature complete Node.js runtime recipes.

I agree that `npm` and Yocto are perfect not match (version/license locking being the main issue), but still do not believe that replacing `npm`'s own dependency and build task handling is safe way forward. If `npm` project gets on board, which they might with their upcoming features, I'd be happy to use that both in Yocto and `meta-nodejs` classes. Currently I see that shrinkwrapping, shipping with bundled npm packages and custom registries are their methods of solving this.

At least I hope that `meta-nodejs` offers alternative way for those who need to be "bit more Node.js" and "less Yocto" with their recipes.

imyller, 2016-05-12:
btw. 

I'm leaning towards renaming our `npm` class to `npm-base`. This seems to be the most conforming and best descriptive way to match existing Yocto classes providing similar helper functions.

 `npm-base` will offer `oe_runnpm` which is guaranteed to be proper way to call `npm` with full cross-compiling capability (for V8 native API modules) and is isolated to recipe workdir in its operating scope. One of the main focuses of oe_runnpm is (and should be) ensuring that there is no npm originated cross-talk between different recipes build tasks. This includes separating npm fetcher caches etc.

`npm-install` will keep its name and will still closely match Yocto 2.1's `npm` class in functionality. It will offer configurable install, dedupe and shrinkwrap build tasks for the recipe. It also supports building native packages, SDKs and globally installed modules (as they are called in npm terminology).

All this "collision avoidance" has to be documented properly and highlighted with the pros and cons for developers wishing to use either Yocto `npm` class or our `npm-base`, `npm-install`  etc. helper classes.

imyller, 2016-05-18:
Following changes have taken place:

* There are now separate branches for: Yocto 2.0 (jethro), Yocto 2.1 (krogoth), master (for staging)
* For Yocto 2.1 krogoth `npm` class has been renamed to `npm-base` (unfortunate, but mandatory)

Recipes dependending on `oe_runnpm` function should be update to inherit `npm-base` instead of `npm`. 

Rationale for choosing `npm-base` as a class name is the fact that Yocto `oe_runmake` is defined in `base` class; and now `oe_runnpm` is defined in `npm-base` class. Both devtools fall into similar category.

At least for now, I'm not bringing this class rename change to `jethro` branch.

bachp, 2016-05-18:
To ease the transition from jethro to krogoth it might be good to add the `npm-base.bbclass` already there so people can already port their recipe.

Maybe the old meta-nodejs `npm.bbclass` could just be a shim that does nothing else then inherit the new `npm-base.bbclass`.

imyller, 2016-05-18:
Great idea. Implemented.

imyller, 2016-05-19:
Closing this issue as fixed with Yocto 2.1 (krogoth) branch

closed 2016-05-19 by imyller