imyller/meta-nodejs #43: Unable to build node-red

graugans, 2016-06-12:
I tried to build node-red with your nodejs layer but I could not succeed. The following recipe was used:

```
DESCRIPTION = "Node-RED"
HOMEPAGE = "http://nodered.org"
LICENSE = "Apache-2.0"
LIC_FILES_CHKSUM = "file://LICENSE;md5=2ee41112a44fe7014dce33e26468ba93"
DEPENDS += " nodejs"
RDEPENDS_${PN} += " nodejs bash python zsh"

inherit npm-install

SRC_URI = "https://github.com/${PN}/${PN}/releases/download/${PV}/${PN}-${PV}.zip \
"

SRC_URI[md5sum] = "442c28584892617bbf300456bc6f1ac9"
SRC_URI[sha256sum] = "badddaebb0ebc9cdc47372b88050a83117b55fb28437f58c5696c5669261db1e"

S = "${WORKDIR}/${PN}-${PV}"

NODE_MODULES_DIR = "${prefix}/lib/node_modules/"

do_install() {
    install -d ${D}${NODE_MODULES_DIR}${PN}
    cp -r ${S}/* ${D}${NODE_MODULES_DIR}${PN}
}

INHIBIT_PACKAGE_DEBUG_SPLIT = "1"

FILES_${PN} = "${NODE_MODULES_DIR}${PN} \
"
```

But I always get the folloeing QA error looks like an architecture missmatch...
```
ERROR: node-red-0.13.4-r0 do_package: runstrip: ''arm-burr-linux-gnueabi-strip' --remove-section=.comment --remove-section=.note '/build/udoo-iot-bsp/build/tmp/work/cortexa9hf-neon-burr-linux-gnueabi/node-red/0.13.4-r0/package/usr/lib/node_modules/node-red/node_modules/node-sass/vendor/linux-x64-48/binding.node'' strip command failed with 1 (arm-burr-linux-gnueabi-strip:/build/udoo-iot-bsp/build/tmp/work/cortexa9hf-neon-burr-linux-gnueabi/node-red/0.13.4-r0/package/usr/lib/node_modules/node-red/node_modules/node-sass/vendor/linux-x64-48/binding.node: File format not recognized
)
ERROR: node-red-0.13.4-r0 do_package_qa: QA Issue: Architecture did not match (40 to 62) on work/cortexa9hf-neon-burr-linux-gnueabi/node-red/0.13.4-r0/packages-split/node-red/usr/lib/node_modules/node-red/node_modules/node-sass/vendor/linux-x64-48/binding.node [arch]
WARNING: node-red-0.13.4-r0 do_package_qa: QA Issue: /usr/lib/node_modules/node-red/node_modules/grunt-cli/completion/zsh contained in package node-red requires /bin/zsh, but no providers found in RDEPENDS_node-red? [file-rdeps]
ERROR: node-red-0.13.4-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: node-red-0.13.4-r0 do_package_qa: Function failed: do_package_qa
ERROR: Logfile of failure stored in: /build/udoo-iot-bsp/build/tmp/work/cortexa9hf-neon-burr-linux-gnueabi/node-red/0.13.4-r0/temp/log.do_package_qa.29079
ERROR: Task 10 (/build/udoo-iot-bsp/sources/meta-burr/recipes-nodejs/node-red/node-red_0.13.4.bb, do_package_qa) failed with exit code '1'
```

This is the output of my bitbake build configuration. I am using ``krogoth`` branch
```
Build Configuration:
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-15.04"
TARGET_SYS        = "arm-burr-linux-gnueabi"
MACHINE           = "udooqdl"
DISTRO            = "burr"
DISTRO_VERSION    = "1.0.0"
TUNE_FEATURES     = "arm armv7a vfp thumb neon       callconvention-hard       cortexa9"
TARGET_FPU        = "hard"
meta              
meta-poky         = "HEAD:8f51f6153a09f8048fb4c4ce9cf4a19655240de4"
meta-oe           
meta-multimedia   
meta-python       
meta-networking   = "HEAD:247b1267bbe95719cd4877d2d3cfbaf2a2f4865a"
meta-fsl-arm      = "HEAD:3f34509381092a893b8efe4451018b8d67a5abf6"
meta-fsl-arm-extra = "HEAD:698cf389aa446244a39b597fbf58be1d55096500"
meta-fsl-demos    = "HEAD:7e4237db1f48cbbde776b17a66d71def14e3809e"
meta-nodejs       = "krogoth:6c6a085e14957e56819f8d7b69ee5b43849d21f5"
meta-burr         = "krogoth:299cacabbb6ec6f0916d14b6afa17282aaf546cf"
meta-udoo         = "HEAD:f01b46b5ed0906eacb2024e0eb110accce1d32c1"
```

imyller, 2016-06-14:
You might want to experiment with adding `INHIBIT_PACKAGE_STRIP = "1"`

graugans, 2016-06-15:
I already tried ``INHIBIT_PACKAGE_STRIP = "1"`` the ``install`` function was added after I received empty ``.ipk`` packages. After digging deeper it looks like a version ``v6.x`` is pulled in during build of ``node-red`` what causes the ``nm`` install to fail. I'll post some logs when back at home.

In my distro conf I pinned the node version to 4.4.5
```
PREFERRED_PROVIDER_node = "nodejs"
PREFERRED_PROVIDER_node-native = "nodejs-native"
PREFERRED_VERSION_nodejs ?= "4.4.5"
```
When I build ``node`` bitbake pulls and builds 4.4.5

```
bitbake node
```
Are there any simple example recipes for a node-js module?

imyller, 2016-06-15:
I'm preparing to publish a separate `meta-nodejs-extras` layer that will include helper classes and Node.js based software.

Few notes:

* I'd suggest you make your `PREFERRED_VERSION_` pinning to version `4` rather than specific major.minor.patch release version. The `4.4.5` recipe file is going away as soon as patch is released by upstream.

*  `PREFERRED_PROVIDER_` should not be necessary anymore since io.js is gone from layer recipes.

* `npm-install` brings its own dependency to `node-native` (for toolchain) so you do not need to `DEPEND` on Node.js if you do not want to.

* Do not forget to add `PREFERRED_VERSION_nodejs-native`

I'll try to take some time to test with node-red and your recipe soon.

imyller, 2016-07-07:
Working recipe for Node-RED can now be found in `meta-nodes-contrib` layer:

https://github.com/imyller/meta-nodejs-contrib

I'll try to keep the recipe in sync with upstream releases of Node-RED.

I'm closing this as resolved.

closed 2016-07-07 by imyller