imyller/meta-nodejs #44: meta-nodejs error with Yocto

rtvs16, 2016-06-14:
I tried putting the layer as indicated and the system indicated an error during compiling.

**Details**:
Using Yocto-fido
Build Configuration:
BB_VERSION        = "1.26.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "beaglebone"
DISTRO            = "poky"
DISTRO_VERSION    = "1.8.2"
TUNE_FEATURES     = "arm armv7a vfp neon callconvention-hard cortexa8"
TARGET_FPU        = "vfp-neon"
meta
meta-yocto
meta-yocto-bsp
meta-nodejs       = "fido:900d7d6b59c36b2bdbd1c85febec99e80ab54f95"

I add nodejs in the local.conf as follows
`IMAGE_INSTALL_append = "nodejs"`

Then I run bitbake nodejs and the system report the next error
`...
make[1]: execvp: printf: Argument list too long`

I gave a try with Yocto-Krogoth  with same results

I saw that this issue was patched before
[https://github.com/imyller/meta-nodejs/issues/9](url)  but it seems that is back, is that the case?

imyller, 2016-06-14:
I noticed that you have
```
meta-nodejs = "fido:900d7d6b59c36b2bdbd1c85febec99e80ab54f95"
```

This repository does not have `fido` branch or even commit `900d7d6b59c36b2bdbd1c85febec99e80ab54f95`

What is the source for your `meta-nodejs` layer?

imyller, 2016-06-14:
Also, same advice applies as given in issue #9 

* Make sure your default shell is `bash`
* Make your layer path as short as possible (OpenSSL is sensitive for this)
* If that does not help, use shared OpenSSL library

rtvs16, 2016-06-14:
Thanks for your quick response.  I am testing in a local copy, that's why you don't see it in the main branch.  I just downloaded your code and install it.

In regards of the bash, I am on that, so it is ok.

But here is the interesting part.  I have one of your  prev version (I beleive) BUT I am using it only in a recipe (instead of a layer) and it compiles and works without a problem.  
Looking at the differences, I can see minor.  Comments?

[nodejs_diff.txt](https://github.com/imyller/meta-nodejs/files/314782/nodejs_diff.txt)
Left side is the old version that works.

imyller, 2016-06-14:
What Node.js version are you trying to build?

rtvs16, 2016-06-14:
0.12

imyller, 2016-06-14:
There have been OpenSSL compilation issues with Node.js 0.x outside Yocto projects too.

I'm currently testing 0.12 builds in my build server.

imyller, 2016-06-14:
Please try with latest repo version for `master` or `krogoth` branch. I'm not maintaining branch for Yocto fido.

`0.12` and `0.8` both compiled successfully with:

```
Build Configuration:
BB_VERSION        = "1.31.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "arm-angstrom-linux-gnueabi"
MACHINE           = "beaglebone"
DISTRO            = "angstrom"
DISTRO_VERSION    = "v2016.06"
TUNE_FEATURES     = "arm armv7a vfp thumb neon       callconvention-hard"
TARGET_FPU        = "hard"
```

rtvs16, 2016-06-15:
Hurray!, It works!  I tested back in fido and it works too!
Thanks!!!

imyller, 2016-06-15:
Glad to hear! I'm closing this issue as resolved.

closed 2016-06-15 by imyller