imyller/meta-nodejs #45: How to add nodejs package with a recipe

rtvs16, 2016-06-16:
Hi,

I have been trying to add to my image a nodejs package without success.  I looked around and I couldn't  find an answer, so perhaps this questions could help others.

I am trying to add node-bindings to my image as follows:
Create a file called -> bindings_1.2.1.bb
```
LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://LICENSE-MIT;md5=331c934843c71c28b2bf98046c03eb97"
S = "${WORKDIR}/git"

SRC_URI = "https://github.com/TooTallNate/node-bindings/archive/v${PV}.tar.gz"

SRC_URI[md5sum] = "b884aeba4283309445a8e3b6e7322dd6"
SRC_URI[sha256sum] = "2d708fb3c024b9e6e86df141faff802194f5db90a4b79e6d4aa6bd61d$

# This package does not need to bee compiled nor configure
do_configure[noexec] = "1"
do_compile[noexec] = "1"

inherit npm-install-global

NPM_INSTALL = "bindings"
```
The log reads
```
npm ERR! errno -21
npm ERR! eisdir EISDIR, read
npm ERR! eisdir This is most likely not a problem with npm itself
npm ERR! eisdir and is related to npm not being able to find a package.json in
npm ERR! eisdir a package you are trying to install.
npm verb exit [ -21, true ]
```
Could you provide some hints / directions on how to install a package / write a recipe ?

imyller, 2016-06-16:
I'll give you an example bb-file in a moment, but first I'll have to ask:

Why are you packaging node-bindings to a globally installable OE package?

Why I am asking:

Usually `npm-install` is used for complete application that gets bundled and packaged with all its node_modules cross-compiled and included for embedded deployment. Another good examples are command line utilities such as `grunt-cli`, `bower` or `forever`.

Packaging each and every dependency module separately is not what I'd recommend for deploying Node.js based software.

imyller, 2016-06-16:
It's this simple:

`node-bindings_1.2.1.bb`
```bitbake
SUMMARY = "Helper module for loading your native module's .node file"
LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://README.md;beginline=73;md5=c9dbec5380fde616b6393b2b1d91cd18"

SRC_URI = "https://github.com/TooTallNate/node-bindings/archive/v${PV}.tar.gz"

SRC_URI[md5sum] = "f2a1591fc44434405f8e5c5622ab8d37"
SRC_URI[sha256sum] = "1fe30d614f6971cc5ab1ecee0edbf95d96d3abe9386a71f5fcad9fcc15bc3c24"

inherit npm-install-global
```

rtvs16, 2016-06-16:
First thank you very much for your help.
In regards of your question, the motivation is to provide an out-of-the-box set with the necessary modules and not worrying about 2d or 3rd installations. 
 I agree with you that using npm is another option (and probably better), nevertheless I see better responses from the users  when the packages are included.

I gave a try to your suggestion and unfortunately did not work .  It is showing similar errors of what I had before

```
npm ERR! code ENOPACKAGEJSON
npm ERR! errno -2  
npm ERR! package.json This is most likely not a problem with npm itself.
npm ERR! package.json npm can't find a package.json file in your current directory.
npm verb exit [ -2, true ]
 ...
ERROR: oe_runnpm failed
```
**Note**:  
Installing the package using npm on the target works fine.  The fine print is that the environmental variable **NODE_PATH** is not defined, so the search for modules sometimes fails.Tthis situation can be corrected adding it to _/etc/environment_

imyller, 2016-06-16:
Did you really replace your recipe with the one that I provided?

Reason your original recipe did not work was that it had `S = "${WORKDIR}/git` which pointed the `S` to wrong directory.

Also you skipped `configure` step and `compile` step for no reason. 

I was able to build proper `node-bindings` package with above recipe.

rtvs16, 2016-06-16:
Yes  I did, it is exactly as yours in content.

I figured it out.... the recipe name had a missing s
mine -> node-binding_1.2.1.bb
yours -> node-binding**s**_1.2.1.bb

My bad on that, sorry;  and again thanks for your help!

imyller, 2016-06-16:
Glad the issue resolved. I named the recipe after original node module.

Closing this as resolved. 

I'm soon releasing a separate layer for Node.js bb recipes which will include lot of examples.

closed 2016-06-16 by imyller

binary-nerd, 2017-09-26:
Hi @imyller 

> I'm soon releasing a separate layer for Node.js bb recipes which will include lot of examples.

Did you release them? Can you please share the appropriate link?

I have few related questions. Could you kindly consider  answering them?

- I tried searching a lot for `npm-install-global` but could not find it. It is not defined in npm.class. Where is it defined?

- If I am just trying to install from package.json on my local machine, how can i modify the script

Thank you very much.

Alex-dev02, 2022-04-05:
Hi @imyller , sorry for bothering, but I have a question. What do I need to do in order to use ```inherit npm-install-global```? I searched the internet but couldn't find anything useful. I tried inheriting it inside poky, to see if it somehow exists inside it, but I keep getting  "Could not inherit file classes/npm-install-global.bbclass".