imyller/meta-nodejs #51: Make shared OpenSSL the default

bachp, 2016-08-17:
Currently nodejs builds by default with the internal OpenSSL. From an OE point I think it would be better to use the the shared openssl by default.

This would also solve the path to long issues.

Maybe the same is true for shared zlib?

imyller, 2016-08-17:
I've been thinking the same thing myself. 👍 for this effort.

I agree that using shared OpenSSL would be better for OE builds. However, I'd still keep zlib internal. It's trivial in size and having Node.js maintain it's own patch sets and integration for the zlib feels more safe choice. zlib has never caused any issues.

I've been following the work around Node.js core and it's integration with zlib and especially OpenSSL. Node.js is very specific about the OpenSSL version used and while compiling with shared OpenSSL is allowed, it's definitely not recommended.

We have to be careful when defining OpenSSL dependency so that too obviously incompatible version wont be accidentally used; if such limitation is reasonably possible. Using incompatible shared OpenSSL could cause catastrophic and very hard to diagnose bugs for end users.

bachp, 2016-08-19:
I don't think we can guarantee the correct OpenSSL version in OE other than using the correct branch. (krogoth<->krogoth).

I just checkend and for example Debian seems to be using external OpenSSL and zlib for their [nodejs](https://packages.debian.org/jessie/nodejs) package too.

We are also using it with shared zlib and openssl via the PACKAGECONFIG without problems on the krogoth branch.

But I understand your concern and I can live with the PACKAGECONFIG options that are already available

imyller, 2016-08-19:
I'd say we make the switch and see how it goes?

Default to shared libs.

I'll test for jethro as I want to keep supporting it for a while without branching too far from krogoth/master.