imyller/meta-nodejs #54: Example of cross compiling module

ringtailsoftware, 2016-10-05:
Is there an example of building a node module where cross compilation of a binary library is performed?

I am trying to build https://github.com/sandeepmistry/node-bluetooth-hci-socket, but gyp seems to be calling the host compiler.

imyller, 2016-10-06:
At least one recipe at [meta-nodejs-contrib](https://github.com/imyller/meta-nodejs-contrib) is cross-compiling native dependencies:

https://github.com/imyller/meta-nodejs-contrib/blob/master/recipes-extended/node-red/node-red_0.14.6.bb

It has dependency to `node-serialport`, which gets cross-compiled as native addon.

imyller, 2016-10-06:
Also, I'd like to remind that cross-compiling `npm` is available only as `oe_runnpm` when you have inherited the `npm-base` class. That is analogous to `oe_runmake` available from `base` class in Yocto, where calling plain `make` in the recipe skips all cross-compilation magic.

If you run `npm` command as-is, you're very likely calling host machine `npm` or the target `npm` with host machine `node` binary - and cross-compilation does not take place.

Helper classes for Node.js recipes such as `npm-install` or `npm-install-global` use `oe_runnpm` internally. This is the case with `node-red` recipe example I gave before.

closed 2016-10-19 by imyller

soonsebii, 2016-11-30:
@ringtailsoftware I hope these help.
https://github.com/soonsebii/meta-nodejs-contrib/blob/master/recipes-extended/noble/noble_1.7.0.bb