imyller/meta-nodejs #55: Compile v6.8.1 on armv7

joaopaulob, 2016-10-18:
Hi,

I'm trying to compile the version v6.8.1 but I'm getting the following warning:

```
NOTE: preferred version 6.8.1 of nodejs not available (for item nodejs)
NOTE: versions of nodejs available: 0.12.7
NOTE: Resolving any missing task queue dependencies
NOTE: preferred version 6.8.1 of nodejs not available (for item nodejs-dev)
NOTE: versions of nodejs available: 0.12.7
NOTE: preferred version 6.8.1 of nodejs not available (for item nodejs)
NOTE: versions of nodejs available: 0.12.7
```

My local.conf:

`PREFERRED_VERSION_nodejs = "6.8.1"`

Build configuration:

`BB_VERSION        = "1.28.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "colibri-imx7"
DISTRO            = "poky"
DISTRO_VERSION    = "2.0.2"
TUNE_FEATURES     = "arm armv7a vfp thumb neon callconvention-hard cortexa7"
TARGET_FPU        = "vfp-neon"
meta              
meta-yocto        = "jethro:37eb21b2b1d5977a028a448e52551607fae99bf7"
meta-oe           
meta-python       
meta-networking   
meta-gnome        
meta-ruby         = "jethro:8ab04afbffb4bc5184cfe0655049de6f44269990"
meta-yocto-bsp    = "jethro:37eb21b2b1d5977a028a448e52551607fae99bf7"
meta-toradex      = "V2.6:485643678b2d39d37b1cc9d1aa2200bb934b08de"
meta-fsl-arm      = "jethro:417f669e4dede244a81e11aa6d8b60c4e333e589"
meta-fsl-arm-extra = "master:fddc5aef6712b9ea87095772ad51a8d41be9ad67"
`

Am i missing anything? I'm using the latest master.

Best regards

gizero, 2016-10-18:
Maybe meta-nodejs is completely missing from your bblayers.conf? I don't see in your config...

imyller, 2016-10-18:
I agree with @gizero 

Looks like meta-nodejs is not included.

joaopaulob, 2016-10-18:
Hi,

My bblayers.conf:

`LCONF_VERSION = "6"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  ${TOPDIR}/../meta \
  ${TOPDIR}/../meta-yocto \
  ${TOPDIR}/../meta-openembedded/meta-oe \
  ${TOPDIR}/../meta-openembedded/meta-python \
  ${TOPDIR}/../meta-openembedded/meta-networking \
  ${TOPDIR}/../meta-openembedded/meta-gnome \
  ${TOPDIR}/../meta-openembedded/meta-ruby \
  ${TOPDIR}/../meta-yocto-bsp \
  ${TOPDIR}/../meta-toradex \ 
  ${TOPDIR}/../meta-fsl-arm \ 
  ${TOPDIR}/../meta-fsl-arm-extra \
  ${TOPDIR}/../meta-browser \
  ${TOPDIR}/../meta-mytapp-chromium \
  "
BBLAYERS_NON_REMOVABLE ?= " \
  ${TOPDIR}/../meta \
  ${TOPDIR}/../meta-yocto \
  "
EXTRALAYERS +=" \
  ${TOPDIR}/../meta-nodejs \
  ${TOPDIR}/../meta-nodejs-contrib \
  "`

Regards

joaopaulob, 2016-10-18:
Found the issue. The layer should not be on EXTRALAYERS section.

Best regards

gizero, 2016-10-20:
Take a look at #9... seems related. I also experienced a similar issue when building from unusual build paths and suspected this was related to the length of the paths

imyller, 2016-10-29:
@joaopaulob 

If bundled openssl compilation fails you can build Node.js with use shared openssl.

This can be enabled with `PACKAGECONFIG`:

http://www.yoctoproject.org/docs/2.1/ref-manual/ref-manual.html#var-PACKAGECONFIG

For example, in your `local.conf`
```
PACKAGECONFIG_append_pn-nodejs = " openssl"
```

or in `nodejs_%.bbappend`:

```
PACKAGECONFIG_append = " openssl"
```

lewicki-pk, 2016-11-02:
using `PACKAGECONFIG_append = " openssl"` helps you because it shortens the command used to link openssl, but it does not touch the core of the issue.

I give you my example. I keep my yocto related stuff in one directory, where I have 2 other dirs: `build` and `layers`. This should be self-explainatory.

If my main path is:

- `/home/myusername12/yocto-builds/2016-11-02` - the build **fails** with `execvp: printf: Argument list too long` error, but if i change it to:

- `/home/myusername12/2016-11-02` - the build **succeeds** (username changed, but directory name length preserved).

In my opinion this issue should be forwarded to the developers responsible for creating nodejs.

There has already been a [patch for this issue](https://patchwork.openembedded.org/patch/72811/) in nodejs 0.8.18 but I tried applying it manually (normal patching failed), but even that did not help.

There is already an issue for it in here: https://github.com/nodejs/node/issues/9137