imyller/meta-nodejs #56: recipe: Node.js v7 cross-compilation dependency issues

imyller, 2016-10-29:
Node.js v7 ships with new version of V8 Javascript engine that requires *native* (host) compilation of  `mkpeephole` utility. `mkpeephole` is fundamental to the design of the new ignition interpreter in V8.

Unfortunately V8 mandates building `mkpeephole` tool with **32-bit** (!) native toolchain.  In addition, if opting to build against external openssl and/or zlib, the compilation requires 32-bit versions of those as well for native system.

Installing `g++-multilib` package to host system allows building Node.js v7 recipes for most embedded targets, but such host system dependency is not included by default in Yocto 2.1 build system requirements.

Different options for resolving this both in Node.js+V8 upstreams and within OpenEmbedded tools are currently being explored. Good ideas are welcome!

This issue has been opened as proper place to continue the discussion started in issue comments:

https://github.com/imyller/meta-nodejs/commit/c54ce2afe0accf84206a61ea8d7954bcb1adc13d#commitcomment-19620853

imyller, 2016-10-29:
/cc @bachp @gizero @ofrobots

imyller, 2016-10-29:
Documented Node v7 requirements to master branch README in commit  https://github.com/imyller/meta-nodejs/commit/0b66df51c1023bb7d2506aee16394774d79ca38a

Hopefully we can resolve the issue and remove this doc before merging v7 in `krogoth` branch.

imyller, 2016-11-29:
No progress yet.

Looks like we are going to have to kindly ask Yocto to include `g++-multilib` (in addition to existing `gcc-multilib`) to host/build system package requirements. I'm not too hopeful of that happening though:

http://www.yoctoproject.org/docs/2.3/ref-manual/ref-manual.html#required-packages-for-the-host-development-system

While I hoped not to, I'm now considering merging v7.x Node.js recipes to krogoth and jethro branches with this dependency requirement.

imyller, 2016-12-07:
Node.js v7.x.x is now in `krogoth` branch.

Unfortunately `g++-multilib` is now build system dependency for all Yocto versions when building Node.js v7 or newer.

Nitnelav, 2017-06-04:
Hi, I struggled a bit finding out this dependency was needed.
Could you add it on the README ?

cristina21r, 2018-03-26:
I'm using rocko..what should i do before bitbake core-minimal, in order to manage working with nodejs>=6 and npm in my linux?