imyller/meta-nodejs #58: 'Cannot find module error' error using Nodejs v4.6.1 (jethro)

soonsebii, 2016-11-29:
Something is the matter with this error.

**Try "export NODE_PATH=/usr/lib/node_modules"**
This will help us resolve the issue faster. However, I don't think it is a fundamental solution.

I think there is a problem here. I don't know why '/usr/lib/node_modules' are missing.
```bash
$ pwd
/home/root
$ node
> console.log(global.module)
Module {
  id: '<repl>',
  exports: {},
  parent: undefined,
  filename: null,
  loaded: false,
  children: [],
  paths:
   [ '/home/root/repl/node_modules',
     '/home/root/node_modules',
     '/home/node_modules',
     '/node_modules',
     '/home/root/.node_modules',
     '/home/root/.node_libraries',
     '/usr/lib/node' ] }
```
Do you have any idea?

imyller, 2016-11-29:
Can you please check if path `/usr/lib/node_modules` exist at all in your system?

soonsebii, 2016-11-29:
The path exists in my system.

I have a little experience with OpenWRT. In the case of OpenWRT,  the configuration is as follows.
I'm not sure, but I think the user path is the problem.
```bash
$ pwd
/root
$ node
> console.log(global.module)
Module {
  id: '<repl>',
  exports: {},
  parent: undefined,
  filename: null,
  loaded: false,
  children: [],
  paths:
   [ '/root/repl/node_modules',
     '/root/node_modules',
     '/node_modules',
     '/root/.node_modules',
     '/root/.node_libraries',
     '/usr/lib/node',
     '/usr/lib/node_modules', ] }
```

soonsebii, 2016-11-30:
I resolve the problem with [patch](https://github.com/soonsebii/meta-nodejs/commit/67e489d88335fcfe7ee0dd69a2eb7217cc8f024f).

Thanks for your concern and closing issue.

closed 2016-11-30 by soonsebii