imyller/meta-nodejs #59: How to build specific version nodejs, such as 4.x without npm? How small can the nodejs be using bitbake??

alex2wong, 2016-12-02:
Hello, It is really nice to get involved in bitbake when I want to build a mini nodejs in linux.
I have add **EXTRALAYERS** in oe/build/conf/bblayers.conf as follows:
```
BBLAYERS ?= " \
  /home/linux/Documents/openembedded-core/meta \
  "
EXTRALAYERS +=" \
        /home/linux/Documents/meta-nodejs \
    "
```
And I copyed the **nodejs** dir in **meta-nodejs/recipes-devtools"" to oe/build/meta/recipes-devtools as the BBLAYERS refer to.
Then I start to bitbake nodejs, this time I don't specify any version, cause last time I use "bitbake -b '/nodejs_v4.6'" **which caused  oe runtime failed ...**, Have you ever encountered this problem??

So "bitbake nodejs" cmd start about total 567 task....It is really slow.... 
**Last question:** 
is there any difference between **bibake** nodejs to official tutorial of nodejs with **make** ??
The **node_v0.12.x** under windows can be smaller than **10MB**, can it be under linux?? Thx!

imyller, 2016-12-02:
You should never break up layer by copying parts of it to a different location. A layer has lot of other files (classes, configuration etc.) that are internally referenced.

Please keep `meta-nodejs` layer and all its subdirectories together as-is.

Your `bblayers.conf` allows Bitbake to locate configuration and recipes for various layers - including `meta-nodejs`.

If configured properly, simple `bitbake nodejs` should be enough to build the latest version.

alex2wong, 2016-12-02:
Well, thx very mush. But how long would it take to finish bitbake tasks ??. It seems very slow to do_fetch, do_compile.
After I start bitbake nodejs, several hours have gone, but it still has over 300 tasks to finish.

closed 2018-11-17 by alex2wong