imyller/meta-nodejs #62: Minify assets

purell, 2016-12-21:
Hi,

It's more a question than an issue but I would like to get some advise on how to **minify** public assets (_javascript, css, etc..._) in the **do_install** task of Yocto. I could install npm package that do this but they can only be run on the target host, I want to do that on the Yocto host but for the target files. Could I add a fetch address of executable that do that on my recipe and then call it to minify the target rootfs assets files ?

Ideally I don't want to use grunt or any other automated target script to minify as the ARM processor we are using has not enough power to minify all the assets in a reasonable time at startup.

Best regards,

imyller, 2016-12-21:
I see no reason why you couldn't use grunt or any other automated method for minifying assets when building the package.

You have `nodejs-native` running locally on build host during `do_install` phase which allows running npm, grunt etc. to build assets, install dependencies for packaging into target system. Also, `oe_runnpm` wrapper for calling npm is able to cross-compile native addons for target system.

Rarely there is need to ship these build-time utilities or package managers (npm, grunt etc.) to low-resource target systems. This is the reason why separate installable packages are built for `nodejs` and `nodejs-npm`.

purell, 2016-12-22:
Hi imyller,

Thank you for you help. Forgive me for my dumb questions.... but after reading on some forum or meta node readme I'm not sure to understand. 

So you telling me that it is possible to use a NATIVE nodejs module through Yocto ? For example, how could I use [node-minify](https://www.npmjs.com/package/node-minify) ?

Here is an basic recipe, I put my questions in the **comments** :
``` 
DEPENDS = "nodejs-native "
RDEPENDS_${PN} = "nodejs "

inherit npm npm-install

do_install() {

        # This build/install dependency for target ?
	oe_runnpm install
	oe_runnpm prune --production

        # Is it the proper way to install node-minify on the host like this ?
        oe_runnpm_native install https://github.com/srod/node-minify/tarball/2.0.2

        # Now, how can I run natively to minify the assets present in the work folder of the recipe ?
}
``` 

Best regards,

purell, 2016-12-22:
Is this better ?

```
do_install() {

        # This build/install dependency for target
	oe_runnpm install
	oe_runnpm prune --production

        # Is it the proper way to install uglify on the build machine  ?
        oe_runnpm_native install uglify

        # Is uglify will be run on the build machine ?
        find . -name "*.js" -exec uglifyjs -o {} {} \;
}
```