imyller/meta-nodejs #69: nodejs-native fails to build on yocto/pyro (Fedora 25)

zonque, 2017-05-04:
On a Fedora 25 box, when building `nodejs-native` with the `pyro` branch of poky, I'm getting the following error.

```
| /tmp/poky/build/tmp-glibc/work/x86_64-linux/nodejs-native/7.9.0-r1.3/node-v7.9.0/out/Release/mkpeephole: error while loading shared libraries: libssl.so.1.0.2: cannot open shared object file: No such file or directory 
| deps/v8/src/v8_base.target.mk:13: recipe for target '/tmp/poky/build/tmp-glibc/work/x86_64-linux/nodejs-native/7.9.0-r1.3/node-v7.9.0/out/Release/obj.target/v8_base/geni/bytecode-peephole-table.cc' failed
| make[1]: *** [/tmp/poky/build/tmp-glibc/work/x86_64-linux/nodejs-native/7.9.0-r1.3/node-v7.9.0/out/Release/obj.target/v8_base/geni/bytecode-peephole-table.cc] Error 127
| make[1]: *** Waiting for unfinished jobs....
| /tmp/poky/build/tmp-glibc/work/x86_64-linux/nodejs-native/7.9.0-r1.3/node-v7.9.0/out/Release/mkpeephole: error while loading shared libraries: libssl.so.1.0.2: cannot open shared object file: No such file or directory 
| deps/v8/src/v8_base.host.mk:13: recipe for target '/tmp/poky/build/tmp-glibc/work/x86_64-linux/nodejs-native/7.9.0-r1.3/node-v7.9.0/out/Release/obj.host/v8_base/geni/bytecode-peephole-table.cc' failed
| make[1]: *** [/tmp/poky/build/tmp-glibc/work/x86_64-linux/nodejs-native/7.9.0-r1.3/node-v7.9.0/out/Release/obj.host/v8_base/geni/bytecode-peephole-table.cc] Error 127
| make[1]: Leaving directory '/tmp/poky/build/tmp-glibc/work/x86_64-linux/nodejs-native/7.9.0-r1.3/node-v7.9.0/out'
| Makefile:75: recipe for target 'node' failed
| make: *** [node] Error 2
| ERROR: Function failed: do_compile (log file is located at /tmp/poky/build/tmp-glibc/work/x86_64-linux/nodejs-native/7.9.0-r1.3/temp/log.do_compile.11837)
```

The cross-compiled version works fine, but something seems to be off with the search path for native libraries. I can fix this but creating a symlink `/usr/lib64/libcrypto.so.1.0.2k` to `/usr/lib64/libcrypto.so.1.0.2` and `/usr/lib64/libssl.so.1.0.2k` to `/usr/lib64/libssl.so.1.0.2` on the host, but that's a hack of course.

zonque, 2017-05-04:
FTR, 7.8 (with d10fd7136 reverted) shows the same issue.

imyller, 2017-05-04:
What is your target arch?

zonque, 2017-05-04:
aarch64

imyller, 2017-05-04:
Oh, sorry. It should not matter with `nodejs-native` 🤕 

I haven't had a chance to test with Yocto "pyro" yet, so officially this repo supports only morty.

Maybe this is a good time to start hacking towards pyro support in `master` branch 👍

zonque, 2017-05-04:
But as I said - the cross-compiled version works, only `nodejs-native` is affected.

imyller, 2017-05-04:
Generally this looks like yet another `mkpeephole` (and `mksnapshot`} build phase issue.

To test a theory, can you modify `nodejs_7.inc` and remove following line in `do_configure`:

```
                --cross-compiling \
```

And try building `nodejs-native` again. Use up-to-date meta-nodejs `master` branch.

If it works, I think this can be fixed easily.

imyller, 2017-05-04:
Also, check if having these packages installed on build host helps:

```
libssl-dev 
libcrypto++-dev 
zlib1g-dev
```

(hopefully these are available in Fedora 25?)

imyller, 2017-05-04:
Just updated. You might want to test again.
* Improved cross-compilation configure flag setting for native builds at 0d6d359d0eef1d283f54e08e5050688876c3500b
* Node.js v7.10.0

zonque, 2017-05-04:
Yes, this works. Perfect, thanks a lot for the quick solution :)

closed 2017-05-04 by zonque