imyller/meta-nodejs #71: Pyro Branch

mdavis777, 2017-05-15:
[Pyro](https://www.yoctoproject.org/downloads/core/pyro23) has finally made it out.
Is it time to create a branch for it?

pabigot, 2017-07-09:
+1 and see pull request #76.

aaronovz1, 2017-08-18:
I'd also like to see a Pyro branch as well as latest Node/NPM recipes. Thanks!

UpsilonAudio, 2017-08-29:
A Pyro branch will also be great for me, thanks :+1:

mdavis777, 2017-08-29:
I think he may be taking a well deserved break / too busy for updates at the moment.
If I get free time I will start pulling changes and fixes into my fork so they are all together and working when he returns.

mdavis777, 2017-08-31:
Started merging all the various outstanding fixes and updates into my fork until he returns.
https://github.com/mdavis777/meta-nodejs

Pyro is branched off there.

UpsilonAudio, 2017-09-12:
Great !
Will try your fork today, will let you know if I see any issues