imyller/meta-nodejs #72: QA staticdev

vpetrigo, 2017-06-07:
Hello,

I have met a [package](https://github.com/kmi/node-red-contrib-noble/tree/master/noble) which provides static library during npm build. And bitbake generates QA error during `do_package_qa` task.

Should the *staticdev* rule be added to `INSANE_SKIP_${PN}` variable for **npm-global-install.bbclass** and **npm-install.bbclass**?

bachp, 2017-06-07:
What warning exactly does bitbake give you?

I think a static library should be packaged into the `-staticdev` package as it normaly would be.

vpetrigo, 2017-06-08:
@bachp thank you for your reply. The exact error message I got:

```
ERROR: QA Issue: non -staticdev package contains static .a library: node-red-contrib-noble path '/work/cortexa8hf-neon-poky-linux-gnueabi/node-red-contrib-noble/0.3.0-r0/packages-split/node-red-contrib-noble/usr/lib/node_modules/node-red-contrib-noble/node_modules/usb/build/Release/usb.a'
non -staticdev package contains static .a library: node-red-contrib-noble path '/work/cortexa8hf-neon-poky-linux-gnueabi/node-red-contrib-noble/0.3.0-r0/packages-split/node-red-contrib-noble/usr/lib/node_modules/node-red-contrib-noble/node_modules/usb/build/Release/obj.target/usb.a' [staticdev]
NOTE: Package node-red-contrib-noble-dbg skipping libdir QA test for PACKAGE_DEBUG_SPLIT_STYLE equals debug-file-directory
ERROR: QA run found fatal errors. Please consider fixing them.
DEBUG: Python function do_package_qa finished
ERROR: Function failed: do_package_qa
```
I have added the split by specifying path where static libraries reside in `FILES_${PN}-staticdev` variable in my recipe. But that exact directories were revealed only after I built the recipe for the package in my first post. The result is:
```
FILES_${PN}-staticdev += " \
    ${libdir_native}/node_modules/node-red-contrib-noble/node_modules/usb/build/Release/*/*.a \
    ${libdir_native}/node_modules/node-red-contrib-noble/node_modules/usb/build/Release/*.a \
"
```
Other packages might spread static libs into much more ways though. It might be tedious to find all this stuff, but I may be mistaken.