imyller/meta-nodejs #73: node v8.x.x

ninox92, 2017-06-15:
I'm working on a pull request including node v8.x.x.
But im struggling on how to implement npm 5.x in this.
This because of the error that is created by calling 
```
${NPM} cache clean || die "oe_runnpm failed (cache clean)"
```
cache clean is no longer tollerated by npm and  'npm cache verify' should be used instead.

Any ideas ?

ninox92, 2017-06-15:
I introduced a new variable to replacing the cache clean command.
```
NPM_CACHE_CMD = "verify"
```
```
${NPM} cache ${NPM_CACHE_CMD} || die "oe_runnpm failed (cache ${NPM_CACHE_CMD})"
```

Is there any we we can determine in npm-base what version of NPM we are running ?

mdavis777, 2017-06-16:
You should be able to just run ${NPM} -- version and parse the version from the return.

ninox92, 2017-06-23:
I'm stuck. Got this far:
NPM_SPLIT is not actually splitting....

```sh
NPM_VERSION=$(${NPM} --v)
NPM_SPLIT="${@'$NPM_VERSION'.split('.')}"
bbnote NPM split $NPM_SPLIT

NPM_CACHE_CMD="${@'verify' if '5' == '$NPM_VERSION'.split('.')[0] else 'clean'}"
```

prints: NPM split ['5.0.3']

thisboyiscrazy, 2017-07-02:
@ninox92 I am interested in the how far did you get access to what you did and maybe help?

ninox92, 2017-07-05:
I subimitted a [pull request](https://github.com/imyller/meta-nodejs/pull/75)

thisboyiscrazy, 2017-07-11:
@ninox92 I might be doing something wrong but the pull request seems to install node v7.10.0 and npm 4.2.0

ninox92, 2017-07-11:
Did you set the 
Local.conf
```
PREFERRED_PROVIDER_nodejs="8.1.3"
```
And did you cloned my fork? Or just this repository. Because node 8.x is not yet included in the main repo.

thisboyiscrazy, 2017-07-11:
Yes I clone your fork I believe (https://github.com/ninox92/meta-nodejs)

And added 

```
PREFERRED_PROVIDER_nodejs="8.1.3"
```

To both local.conf and my image.bb

mdavis777, 2017-07-11:
Shouldn't it be 

PREFERRED_VERSION_nodejs = "8.1.3"
or
PREFERRED_VERSION_nodejs = "8.%"

ninox92, 2017-07-11:
@thisboyiscrazy @mdavis777 
My bad, typo.
```
PREFERRED_VERSION_nodejs
```

thisboyiscrazy, 2017-07-11:
I tried putting

```
PREFERRED_VERSION_nodejs = "8.1.3"
```

in both my nj-image.bb and local.conf, still i am only getting node v 7.10.0

thisboyiscrazy, 2017-07-11:
When I put it in local.config I get:

```
NOTE: Resolving any missing task queue dependencies
NOTE: preferred version 8.1.3 of nodejs not available (for item nodejs)
NOTE: versions of nodejs available: 0.10.48 0.12.18 0.8.28 4.8.3 6.10.3 7.10.0 7.9.0
NOTE: preferred version 8.1.3 of nodejs not available (for item nodejs-npm)
NOTE: versions of nodejs available: 0.10.48 0.12.18 0.8.28 4.8.3 6.10.3 7.10.0 7.9.0
NOTE: preferred version 8.1.3 of nodejs not available (for item nodejs-dev)
NOTE: versions of nodejs available: 0.10.48 0.12.18 0.8.28 4.8.3 6.10.3 7.10.0 7.9.0
```

ninox92, 2017-07-11:
If you don't have: 
```
Recipes-devtools/nodejs/nodejs_8.1.3.bb
```
Then you didn't clone my fork.

thisboyiscrazy, 2017-07-11:
It's possible I got it from here https://github.com/ninox92/meta-nodejs (master branch). Is that the correct location? I did not see that file in https://github.com/ninox92/meta-nodejs/tree/master/recipes-devtools/nodejs

thisboyiscrazy, 2017-07-11:
I also don't see that file in the pull request. Can you point me to the correct repo?

ninox92, 2017-07-12:
@thisboyiscrazy somehow they where not commited. Try again now

thisboyiscrazy, 2017-07-12:
@ninox92 It's working now thanks., Just fyi, it looks like npm 5.2.0 is out. It has a nice new program npx.

ninox92, 2017-07-13:
@thisboyiscrazy That is correct, but npm is bundled with node. 
According to the nodejs [website](https://nodejs.org/en/download/current/)
```
Latest Current Version: v8.1.4 (includes npm 5.0.3)
```

ninox92, 2017-07-13:
@this

closed 2017-07-13 by ninox92