imyller/meta-nodejs #82: nodejs_7.10 requires openssl10

rebeccasf, 2017-08-18:
When compiling meta-nodejs with poky (master), I got into compilation error below.
It is a snippet of it, not sure whether it helps, but I'm using bbappend to remove dependency to openssl v1.1 and add dependency to openssl10. This workaround has temporarily resolved the build issue.

| In file included from ../src/node_crypto.h:20:0,
|                  from ../src/node_crypto.cc:3:
| /media/rebeccas/yocto/11451/build/tmp/work/i586-poky-linux/nodejs/7.10.0-r1.4/recipe-sysroot/usr/include/openssl/ssl.h:228:16: note: forward declaration of 'SSL_SESSION {aka struct ssl_session_st}'
|  typedef struct ssl_session_st SSL_SESSION;
|                 ^~~~~~~~~~~~~~
| ../src/node_crypto.cc: In instantiation of 'static int node::crypto::SSLWrap<Base>::SSLCertCallback(SSL*, void*) [with Base = node::crypto::Connection; SSL = ssl_st]':
| ../src/node_crypto.cc:2930:52:   required from here
| ../src/node_crypto.cc:2413:13: error: invalid use of incomplete type 'SSL {aka struct ssl_st}'
|    ocsp = s->tlsext_status_type == TLSEXT_STATUSTYPE_ocsp;
|           ~~~^~~~~~~~~~~~~~~~~~
| In file included from /media/rebeccas/yocto/11451/build/tmp/work/i586-poky-linux/nodejs/7.10.0-r1.4/recipe-sysroot/usr/include/openssl/crypto.h:31:0,
|                  from /media/rebeccas/yocto/11451/build/tmp/work/i586-poky-linux/nodejs/7.10.0-r1.4/recipe-sysroot/usr/include/openssl/comp.h:16,
|                  from /media/rebeccas/yocto/11451/build/tmp/work/i586-poky-linux/nodejs/7.10.0-r1.4/recipe-sysroot/usr/include/openssl/ssl.h:47,
|                  from ../src/node_crypto.h:20,
|                  from ../src/node_crypto.cc:3:
| /media/rebeccas/yocto/11451/build/tmp/work/i586-poky-linux/nodejs/7.10.0-r1.4/recipe-sysroot/usr/include/openssl/ossl_typ.h:144:16: note: forward declaration of 'SSL {aka struct ssl_st}'
|  typedef struct ssl_st SSL;
|                 ^~~~~~
| ../src/node_crypto.cc: In instantiation of 'static void node::crypto::SSLWrap<Base>::GetTLSTicket(const v8::FunctionCallbackInfo<v8::Value>&) [with Base = node::crypto::Connection]':
| ../src/node_crypto.cc:1306:22:   required from 'static void node::crypto::SSLWrap<Base>::AddMethods(node::Environment*, v8::Local<v8::FunctionTemplate>) [with Base = node::crypto::Connection]'
| ../src/node_crypto.cc:2721:24:   required from here
| ../src/node_crypto.cc:1879:32: error: invalid use of incomplete type 'SSL_SESSION {aka struct ssl_session_st}'
|    if (sess == nullptr || sess->tlsext_tick == nullptr)
|                           ~~~~~~^~~~~~~~~~~
| In file included from ../src/node_crypto.h:20:0,
|                  from ../src/node_crypto.cc:3:
| /media/rebeccas/yocto/11451/build/tmp/work/i586-poky-linux/nodejs/7.10.0-r1.4/recipe-sysroot/usr/include/openssl/ssl.h:228:16: note: forward declaration of 'SSL_SESSION {aka struct ssl_session_st}'
|  typedef struct ssl_session_st SSL_SESSION;
|                 ^~~~~~~~~~~~~~
| ../src/node_crypto.cc:1884:37: error: invalid use of incomplete type 'SSL_SESSION {aka struct ssl_session_st}'
|        reinterpret_cast<char*>(sess->tlsext_tick),
|                                ~~~~~~^~~~~~~~~~~
| In file included from ../src/node_crypto.h:20:0,
|                  from ../src/node_crypto.cc:3:
| /media/rebeccas/yocto/11451/build/tmp/work/i586-poky-linux/nodejs/7.10.0-r1.4/recipe-sysroot/usr/include/openssl/ssl.h:228:16: note: forward declaration of 'SSL_SESSION {aka struct ssl_session_st}'
|  typedef struct ssl_session_st SSL_SESSION;
|                 ^~~~~~~~~~~~~~
| ../src/node_crypto.cc:1885:13: error: invalid use of incomplete type 'SSL_SESSION {aka struct ssl_session_st}'
|        sess->tlsext_ticklen).ToLocalChecked();
|        ~~~~~~^~~~~~~~~~~~~~
| In file included from ../src/node_crypto.h:20:0,
|                  from ../src/node_crypto.cc:3:
| /media/rebeccas/yocto/11451/build/tmp/work/i586-poky-linux/nodejs/7.10.0-r1.4/recipe-sysroot/usr/include/openssl/ssl.h:228:16: note: forward declaration of 'SSL_SESSION {aka struct ssl_session_st}'
|  typedef struct ssl_session_st SSL_SESSION;
|                 ^~~~~~~~~~~~~~
| ../src/node_crypto.cc: At global scope:
| ../src/node_crypto.cc:195:13: warning: 'void node::crypto::crypto_lock_cb(int, int, const char*, int)' defined but not used [-Wunused-function]
|  static void crypto_lock_cb(int mode, int n, const char* file, int line) {
|              ^~~~~~~~~~~~~~
| ../src/node_crypto.cc:183:13: warning: 'void node::crypto::crypto_threadid_cb(CRYPTO_THREADID*)' defined but not used [-Wunused-function]
|  static void crypto_threadid_cb(CRYPTO_THREADID* tid) {
|              ^~~~~~~~~~~~~~~~~~
| node.target.mk:161: recipe for target '/media/rebeccas/yocto/11451/build/tmp/work/i586-poky-linux/nodejs/7.10.0-r1.4/node-v7.10.0/out/Release/obj.target/node/src/node_crypto.o' failed
| make[1]: *** [/media/rebeccas/yocto/11451/build/tmp/work/i586-poky-linux/nodejs/7.10.0-r1.4/node-v7.10.0/out/Release/obj.target/node/src/node_crypto.o] Error 1
| make[1]: Leaving directory '/media/rebeccas/yocto/11451/build/tmp/work/i586-poky-linux/nodejs/7.10.0-r1.4/node-v7.10.0/out'
| Makefile:75: recipe for target 'node' failed
| make: *** [node] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /media/rebeccas/yocto/11451/build/tmp/work/i586-poky-linux/nodejs/7.10.0-r1.4/temp/log.do_compile.38470)
ERROR: Task (/media/rebeccas/yocto/11451/meta-nodejs/recipes-devtools/nodejs/nodejs_7.10.0.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 718 tasks of which 386 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /media/rebeccas/yocto/11451/meta-nodejs/recipes-devtools/nodejs/nodejs_7.10.0.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

mdavis777, 2017-08-31:
Openembedded has reversed the openssl10 decision at least until the next release(2.5).   This is no longer needed.

guillaumekh, 2018-11-26:
@rebeccasf can you please share that bbappend ?

awwx, 2019-04-02:
I ran into this issue today with Yocto thud.

How should nodejs be configured to use openssl10?

Not knowing what I'm doing, I tried creating a `nodejs_7.10.0.bbappend` file which changed PACKAGECONFIG to use `openssl10` instead of `openssl`:

```
PACKAGECONFIG = "zlib openssl10"
```

This seemed to have worked (it compiled at least!), though I did get this warning:

```
WARNING: nodejs-7.10.0-r1.4 do_configure: QA Issue: nodejs: invalid PACKAGECONFIG: openssl10 [invalid-packageconfig]
```

Is this the preferred or recommended method to get node to compile on Yocto thud?

Thanks!

mdavis777, 2019-04-02:
No that would actually build it without openssl at all.  Your bbappend would need
`PACKAGECONFIG[openssl] = "--shared-openssl,,openssl10,"
`
I will note that unless for some reason you are tied to 7.x openembedded/meta-oe already has node 8.x support for thud.  The master branch has 10.x

awwx, 2019-04-02:
@mdavis777  Thank you!

No, no, I'm not tied to 7.x at all.

How would I build npm packages using Node 8.x from openembedded/meta-oe?  Should I be trying to use meta-nodejs at all?  When I added meta-nodejs so that I could use `npm-install`, it switched to building nodejs 7.10.0 from meta-nodejs (nodejs_7.10.0.bb).  Should I be adjusting my layer priorities so that meta-oe has higher priority than meta-nodejs?

Thanks!

awwx, 2019-04-02:
Oh wait, I had added meta-nodejs using bitbake-layers add instead of following your instructions... let me try again :-)

mdavis777, 2019-04-02:
@awwx 
If you use meta-oe you would drop meta-nodejs entirely, and inhert the npm bbclass.
Should all work fine unless you require offline builds which don't currently work in meta-oe npm.

Some links that are a bit outdated but can get you started.
[Book](https://books.google.com/books?id=6NRJDwAAQBAJ&pg=PA334&lpg=PA334&dq=yocto+recipe+inherit+npm&source=bl&ots=REZbpg6CFP&sig=ACfU3U2B4JYlm_48YFtFRgbp8tcb6Bz5TQ&hl=en&sa=X&ved=2ahUKEwj67Z2r07HhAhUQRqwKHZbyAfo4ChDoATAGegQICRAB#v=onepage&q=yocto%20recipe%20inherit%20npm&f=false)
[Wiki](https://wiki.yoctoproject.org/wiki/TipsAndTricks/NPM)

closed 2023-12-19 by rebeccasf