imyller/meta-nodejs #90: nodejs compile error, nodejs-native compile fine

hallard, 2017-12-23:
Hi there,
I had a problem compiling nodejs for yocto on Ubuntu-16.04 (worked on Debian Stretch)
```
...
...
| /usr/bin/ld: cannot find -lz
| /usr/bin/ld: cannot find -lcrypto
| /usr/bin/ld: cannot find -lssl
...
...
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

Of course I installed the 3 libraries before but now way

I solved the issue changing

```
bitbake nodejs
```
to 
```
bitbake nodejs-native
```
not sure yo understand the differences, but now compile fine ;-)

Any idea of what I've done wrong?

bachp, 2017-12-23:
`nodejs` is the package for the target, `nodejs-native` the one for the host system where the build happens.
It looks like there are some dependencies missing.

Can you give some more information like what version of Yocto and meta-nodejs you are using?

hallard, 2017-12-31:
Sorry my fault, I build for debian without any problem, when I switched to ubuntu  I forgot this one, now works fine
```
sudo apt-get libssl-dev:i386 libcrypto++-dev:i386 zlib1g-dev:i386
```
I was using this procedure for Lorix One LoraWan gateways 
https://github.com/Wifx/meta-wifx

closed 2017-12-31 by hallard