imyller/meta-nodejs #97: Getting Illegal instruction - Building PPC e500

arrathin, 2019-12-16:
I can build node in the node-v8.11.3 version. But while run node.js on PowerPC , I get only error "Illegal instruction".
===================================Run Node JS ========================
(gdb) run
Starting program: /pureNeApp/EC/node
Using host libthread_db library "/lib/tls/libthread_db.so.1".
[Thread debugging using libthread_db enabled]
[New Thread 805427584 (LWP 697)]
[New Thread 813861664 (LWP 700)]
[New Thread 822250272 (LWP 701)]
[New Thread 830638880 (LWP 702)]
[New Thread 839027488 (LWP 703)]

Program received signal SIGILL, Illegal instruction.
[Switching to Thread 805427584 (LWP 697)]
0x2550ffa4 in ?? ()
===================================In GDB=============================
# Fatal error in ../deps/v8/src/ppc/assembler-ppc.h, line 776
# Debug check failed: !ra.is(r0).

Program received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 805427584 (LWP 10184)]
0x124aea9c in ?? ()
======================================================================
CPU info from NE:
===============
# cat /proc/cpuinfo
processor       : 0
cpu             : e500
revision        : 2.0 (pvr 8020 0020)
bogomips        : 524.28
chipset         : 8555E
Vendor          : Motorola SPS
Machine         : CPLAB PQ3MC1
bus freq        : 528.000000 MHz
PVR             : 0x80200020
SVR             : 0x80790011
PLL setting     : 0x4
Memory          : 2044 MB

export ProcessPathArmada="/opt/WR/WindRiverLinux-8.0-190909/ppc-85xx/bin"
export PathSysroot=/opt/MV/401CGE-host.26112018/montavista/cge/devkit/ppc/85xx/target
unset MAKEFLAGS
export PATH=$ProcessPathArmada:$PATH
export CFLAGS="-pipe -g  -fno-caller-saves -I$PathSysroot/usr/include/nptl"
export CXXFLAGS="-pipe -g -std=c++11 -fno-caller-saves -I$PathSysroot/usr/include/nptl"
export LDFLAGS="-L$PathSysroot/usr/lib/nptl"
export CC=$ProcessPathArmada/ppc_85xx-gcc
export CXX=$ProcessPathArmada/ppc_85xx-g++
export CPP=$ProcessPathArmada/ppc_85xx-gcc
export AS=$ProcessPathArmada/ppc_85xx-as
export LD=$ProcessPathArmada/ppc_85xx-ld

Required some assistance to move forward this issue.

arrathin, 2019-12-16:
Steps to build 
./configure --dest-cpu=ppc --without-snapshot --without-intl --partly-static --debug --gdb
$ make -j6 node

arrathin, 2019-12-26:
@Matthew Brandyberry

arrathin, 2019-12-26:
@mbroadst

arrathin, 2019-12-26:
cc @nodejs/platform-ppc

arrathin, 2019-12-26:
> cc @nodejs/platform-ppc

Required some guidence

arrathin, 2019-12-27:
@mbroadst

arrathin, 2020-01-02:
@richardlau

richardlau, 2020-01-02:
I’m not aware that anyone has successfully got Node.js working on an e500 (the V8 engine uses PPC instructions that are not available on that processor). There’s a few people trying in https://github.com/nodejs/node/issues/30323 if that helps.

arrathin, 2020-01-06:
@richardlau 
Thanks for your support. i gone through the link but that' not help much